berkeley-abc/abc #41: Assertion 'Abc_NtkIsStrash(pNtk)' failed in 'write_cnf'

kobpko, 2019-05-20:
I'm trying to convert verilog file to cnf file. I have read the verilog file successfully and can use command 'sat' to solve the miter. But when I use 'write_cnf' to output the cnf file. The abc aborted by throwing the failure: 'abc: src/base/abci/abcDar.c:1812: Abc_NtkDarToCnf: Assertion `Abc_NtkIsStrash(pNtk)' failed.'  Can you give me some advice?

daveshah1, 2019-05-20:
Have you tried running `strash` before `write_cnf`? Usually a run of `strash` fixes this assertion.

kobpko, 2019-05-20:
> Have you tried running `strash` before `write_cnf`? Usually a run of `strash` fixes this assertion.

That really works!  
Thanks!

noopurrkalawatia, 2019-11-27:
I tried converting a .bench file into a .cnf file using the following workflow,

read_cnf AND.bench
strash
write_cnf AND.cnf

The tranformation basically changes the OR gate,
INPUT(pi00)
INPUT(pi01)
OUTPUT(po0)
po0  = OR(pi00, pi01)

Into this following file,

c Result of efficient AIG-to-CNF conversion using package CNF
p cnf 5 5
-2 3 4 0
2 -3 0
2 -4 0
-5 0
2 0

The number of variables in the CNF has increased to 5 and I am not exactly sure which transformation scheme is used here. 
It would be of great help if you could clarify this.