berkeley-abc/abc #42: Segmentation Fault While Trying to Write Verilog

sponge611, 2019-05-27:
I have tried to output the mapping result by "write_verilog", but after I typed the command, the system showed that there was segmentation fault.
I also tried to output the mapping result in blif format, and it works.
I tried to search for what caused the segmentation fault problem, but can't get the information.

matt-mdf, 2020-02-25:
Preload libsegfault to get diagnostics. You may need to install the software (linux or bsd). Then set up LD_PRELOAD=${path/to/libsegfault} and run again.

HackHerz, 2023-07-17:
I do have the same problem, which i reproduced on OS X with clang 14.0.3 and on Ubuntu using gcc 11.3.0. Happens for me when i try to convert the following (or a similar) PLA file to verilog:

```
.i 8
.o 6
.ilb layerAf1Ain0A1 layerAf1Ain0A0 layerAf1Ain1A1 layerAf1Ain1A0 layerAf1Ain2A1 layerAf1Ain2A0 layerAf1Ain3A1 layerAf1Ain3A0
.ob layerAf1Aout0A1 layerAf1Aout0A0 layerAf1Aout1A1 layerAf1Aout1A0 layerAf1Aout2A1 layerAf1Aout2A0
.p 3
01000100 010100
01010100 010101
01010000 111101
.e
```

Then run `abc -q "read_pla test.pla; write_verilog test.v"`

Gdb yields the following output:
[image]