berkeley-abc/abc #47: write_aiger with initialized latches produces wrong output

meyerphi, 2019-07-18:
I save the following circuit with a one-initialized latch and a single output as binary AIGER in `test.aig`:
```
aig 1 0 1 1 0
3 1
2
l0 l0
o0 o0
```
Then I invoke ABC as follows:
```
abc 01> read_aiger test.aig
abc 02> write_aiger -s test.abc.aig
```
The resulting output in `test.abc.aig` is (with stripped comments):
```
aig 1 0 1 0 0 1 0
3 1
2
l0 l0
b0 o0
```
This file uses the extended header format, even though this wasn't specified, and the output `o0` is incorrectly converted to a "bad state" property (see [AIGER 1.9 format](http://fmv.jku.at/papers/BiereHeljankoWieringa-FMV-TR-11-2.pdf)).

I expect the same output as the input file, i.e. a single normal output `o0` and no extended header.

alanminko, 2019-07-18:
AIGER parser in ABC only accepts a binary AIGER and does not accept 
AIGER in the text form. One workaround is to use public AIGER tools to 
convert the text format into the binary format.

alanminko, 2019-07-18:
> The binary format is not the issue here, above files are actually
> valid binary AIGER files (as there are no and gates), and conversion
> with aigtoaig produces the exact same files.

Please try &read and &write instead of "read_aiger" and "writer_aiger".  
It may be that these commands are not fully compliant with the latest 
AIGER specs.

meyerphi, 2019-07-18:
Okay, with `&read` I get the message "Converted 1 1-valued FFs and 0 DC-valued FFs.", and after a `&write` all latches are converted to be zero-initialized. Except for that, the circuits still seem to behave the same after a brief inspection. However the symbols are lost in the output, I do not see an option for `&write` to keep the symbols.

I am actually using ABC to rewrite these circuits with commands such as `balance`, and after `&read` all commands just give "Error: Empty network.". How can I still use them?

This is also still an issue with `write_aiger`. Looking at the source code, the issue seems to be in `src/base/io/ioWriteAiger.c` in the tests at line [282](https://github.com/berkeley-abc/abc/blob/0ca7e355f457363bc78de8468df1e66e61b5e8b0/src/base/io/ioWriteAiger.c#L282) and [657](https://github.com/berkeley-abc/abc/blob/0ca7e355f457363bc78de8468df1e66e61b5e8b0/src/base/io/ioWriteAiger.c#L657), which switches to extended format, in combination with lines [315](https://github.com/berkeley-abc/abc/blob/0ca7e355f457363bc78de8468df1e66e61b5e8b0/src/base/io/ioWriteAiger.c#L315) and [319](https://github.com/berkeley-abc/abc/blob/0ca7e355f457363bc78de8468df1e66e61b5e8b0/src/base/io/ioWriteAiger.c#L319). Deleting the checks in line 282 and 657 (thus allowing compact format with one-initialized latches) produces the right results for me, but I don't know if this breaks anything. It seems the changes were introduced in https://github.com/berkeley-abc/abc/commit/34ab59574ee4feedfd1ffb618e2e71a6a1964a6a but I do not see what the intention of that commit was.

alanminko, 2019-07-18:
> introduced in 34ab595

Please use &put/&get to move the internal AIG from the &-space (where 
the &-commands work) to the main-space (where the rest of the commands 
work) and vice versa.