berkeley-abc/abc #48: Question - Adding SWIG interface to construct AIG network programmatically

rkrishnasanka, 2019-07-22:
Hi Team,

I was thinking of adding a SWIG (http://www.swig.org/) interface to ABC for a project that I'm working on that basically needs to do combinatorial logic synthesis. I was wondering if I could get some help on which classes/methods I would need to use programmatically. It there are any resources/tutorials/implementations I would be really grateful.

alanminko, 2019-07-23:
I am not sure that SWIG will be useful. It depends on what you hare 
trying to do with ABC.  Please feel free to describe this. Also, using 
ABC on the command line does not work for you, please have a look at 
program "demo.c", which shows how to use ABC as a static library.

rkrishnasanka, 2019-07-24:
@alanminko thanks a lot for that pointer, I'll take a look at demo.c and let you know. I'm basically working on an HDL for microfluidic devices (which is similar to Verilog). 

1) I wanted to be able to capture logic descriptions in the HDL and then minimize them to simplified combinatorial logic networks
2) I want to be able to take in boolean functions and then generate a combinatorial logic network that will satisfy it
3) Take in state machine descriptions and reduce that into sequential logic networks

My understanding is that ABC should be able to handle all of these scenarios. Please correct me if I'm wrong.

rkrishnasanka, 2019-07-25:
@alanminko thanks a lot for your inputs. I was looking at `demo.c` and I realized that it seems like the general interface from the static library just uses file inputs as **commands** which dictates the entire execution. In a way, it simplifies how much work needs to be done to make a python interface. I think I'll use `cython` since we can compile a static library with ABC.

So I guess I need to figure out how to perform the 3 things I mentioned above using ABC. Do you have any pointers on what references I could use to do them?

alanminko, 2019-07-25:
Good to know that demo.c may work for you.  Regarding your questions:

> 1. I wanted to be able to capture logic descriptions in the HDL and
>     then minimize them to simplified combinatorial logic networks
>  2. I want to be able to take in boolean functions and then [...]
ABC can do all of these.  However, ABC does not accept RTL Verilog.  One 
may need to parse Verilog using Yosys (http://www.clifford.at/yosys/) 
before giving it to ABC. Alternatively, one can use simpler formats for 
representing combinational logic functions: PLA 
(http://www.ecs.umass.edu/ece/labs/vlsicad/ece667/links/espresso.5.html), 
BLIF (https://www.cse.iitb.ac.in/~supratik/courses/cs226/spr16/blif.pdf) 
or AIGER (http://fmv.jku.at/aiger/).

rkrishnasanka, 2019-07-25:
@alanminko Thanks for the resources. Since I'm making the HDL parser myself so I can probably learn a few tricks from Yosys but I don't think I would want to use it directly. 

Just to further clarify, will ABC read PLA, BLIF, and AIGER?