berkeley-abc/abc #49: Using constraints in verification?

zhanghongce, 2019-07-30:
The output other than the first one can be used for constraints in newer AIGER format , as is used by Yosys when translates `$assume` into AIGER. In `read_aiger` command, parsing of the new header is supported. I wonder if they can be truly used in any verification command in ABC? (I tried and it seemed that `pdr` does not support the extra output as constraints, or I just didn't find the right way...)

alanminko, 2019-07-30:
If I remember correctly, we need to read the design using &read (rather 
than "read_aiger").  Next we can do the following: "&put; fold; pdr".  
This should add the constraints present in the AIGER format to the 
miter, so that "pdr" can use them.

zhanghongce, 2019-08-01:
Thank you very much!

closed 2019-08-01 by zhanghongce