berkeley-abc/abc #53: License?

curtisma, 2019-10-01:
Could you please add a file called "LICENSE" to this repository.

I believe the license can be copied from this location: 
https://people.eecs.berkeley.edu/~alanmi/abc/copyright.htm

rodrigomelo9, 2020-01-14:
The license is already in file [copyright.txt](https://github.com/berkeley-abc/abc/blob/master/copyright.txt)

Apteryks, 2025-05-02:
What about the line `The source code is provided for research and evaluation only. For commercial usage, please contact Prof. Chang Wu at wuchang@fudan.edu.cn` seen at https://github.com/berkeley-abc/abc/blob/75adf123f62de9f0b44412bc1354fc1e43070156/README.md?plain=1#L15; does it have any bearing on the distribution of the source?

Apteryks, 2025-05-02:
Looks OK based on https://directory.fsf.org/wiki/ABC, and it's included in Debian (these days as `yosys-abc`), which is a good sign.  The Debian copyright file containing their verifications is visible at: https://metadata.ftp-master.debian.org/changelogs//main/y/yosys/yosys_0.52-1_copyright

I think this can be closed.