berkeley-abc/abc #55: Assertion using &if:  `pCutSet->nCuts > 0'

daveshah1, 2019-10-26:
A Yosys user reported the following assertion from ABC on their design:

```
yosys-abc: src/map/if/ifMap.c:460: void If_ObjPerformMappingAnd(If_Man_t *, If_Obj_t *, int, int, int): Assertion `pCutSet->nCuts > 0' failed.
```

this is happening inside `&if`

Attached is the abc input, run abc.script to reproduce.
[abc_crash_if.zip](https://github.com/berkeley-abc/abc/files/3774747/abc_crash_if.zip)

daveshah1, 2019-10-26:
This seems like a bit of a hack, and perhaps there is a better approach to calculating required times that will fix this, but the following patch avoids the assertion failure:
```diff
diff --git a/src/map/if/ifMap.c b/src/map/if/ifMap.c
index afeba005..50e14827 100644
--- a/src/map/if/ifMap.c
+++ b/src/map/if/ifMap.c
@@ -444,7 +444,7 @@ void If_ObjPerformMappingAnd( If_Man_t * p, If_Obj_t * pObj, int Mode, int fPrep
             pCut->Delay = If_CutDelay( p, pObj, pCut );
         if ( pCut->Delay == -1 )
             continue;
-        if ( Mode && pCut->Delay > pObj->Required + p->fEpsilon )
+        if ( Mode && pCut->Delay > pObj->Required + p->fEpsilon && pCutSet->nCuts > 0 )
             continue;
         // compute area of the cut (this area may depend on the application specific cost)
         pCut->Area = (Mode == 2)? If_CutAreaDerefed( p, pCut ) : If_CutAreaFlow( p, pCut );
```

alanminko, 2019-10-27:
Thank you for being creative!  The fix is basically saying: when the 
mapper is in delay-mode (Mode==1) and delay of the output of the cut 
exceeds its required times, we will accept it - if there is no other 
cuts.  Indeed, this will prevent the situation when no cuts meet the 
required times (as in this case), but in general it can lead to an 
increase in the delay of the mapping...   Please feel free to use the 
fix until I have time to look into it (likely in a week or two).

davidar, 2024-03-31:
I was encountering this issue, and can confirm that the patch suggested in https://github.com/berkeley-abc/abc/issues/55#issuecomment-546612665 resolves it. It may not be ideal, but it's better than a crash, given that a better solution hasn't been implemented in the 5 years since...