berkeley-abc/abc #56: Shared libary is istalled w/out headers

yurivict, 2019-10-26:
All installed files:
```
$ pkg info -l abc
abc-g20191026:
	/usr/local/bin/abc
	/usr/local/lib/libabc.so
```

Library is useless without headers. Other projects, like ```vtr-verilog-to-routing``` expect abc's headers.

yurivict, 2019-10-26:
Rather: both cmake and gmake don't install anything.

yurivict, 2019-10-26:
cmake doesn't even build the target ```libabc-pic```.

yurivict, 2019-10-26:
Please just build one shared library, ```libabc.so```, with position-independenct code. when you install it, please make sure that it has a SONAME variant like ```lib/libabc.so.1.0.0```.

sterin, 2019-10-28:
We were not involved with and do not maintain the FreeBSD abc package.

The CMake build system is an ugly workaround for use in another tool (it runs the regular Makefile and extracts options from it). It is not officially supported.

If there's change or feature you need from us in order to be able to produce a better package for ABC, please explain in more detail what you want us to do and why it is needed. For example, I can add an option to the Makefile to set the SONAME, but you'll need to tell me what options to use and help us test it as we do not have a FreeBSD system easily available.

yurivict, 2019-10-28:
You simply need to add installation instructions for headers into the CMakeLists.txt file.