berkeley-abc/abc #59: Which transformation is used in order to convert a bench file into CNF

noopurrkalawatia, 2019-11-27:
I tried converting a .bench file into a .cnf file using the following workflow,

read_cnf AND.bench
strash
write_cnf AND.cnf

The tranformation basically changes the OR gate,
INPUT(pi00)
INPUT(pi01)
OUTPUT(po0)
po0 = OR(pi00, pi01)

Into this following file,

c Result of efficient AIG-to-CNF conversion using package CNF
p cnf 5 5
-2 3 4 0
2 -3 0
2 -4 0
-5 0
2 0

The number of variables in the CNF has increased to 5 and I am not exactly sure which transformation scheme is used here.
It would be of great help if you could clarify this.

alanminko, 2019-11-27:
It looks like variable "2" represents the output (po0).  It is defined 
in terms of primary inputs (pi00 and pi01) and asserted to be 1 (clause 
"2 0").  Variable "5" represents a constant 0 node, which is not used in 
this circuit, but created by default. Variable "1" is not used in the 
CNF, but still counted.  Not sure what it is reserved for.  This is an 
old CNF generator.  It is recommended to use a better one:

read <file>; strash; &get; &write_cnf -K 8 <out.cnf>

We can also optionally synthesize the circuit using "dc2" or "rwsat" 
after "strash" and before "&get".

ashwath129, 2020-07-07:
> I tried converting a .bench file into a .cnf file using the following workflow,
>
> read_cnf AND.bench
> strash
> write_cnf AND.cnf
>
> The tranformation basically changes the OR gate,
> INPUT(pi00)
> INPUT(pi01) [...]

Where you able to clarify this ? Why the said increase in number of variables and clauses ? I am kind of having the same issue

sterin, 2020-07-10:
Variables 3 and 4 correspond to `pi00` and `pi01`, while variable 2 represents the`po0`.

The clauses 

```
-2 3 4 0
2 -3 0
2 -4 0
```

represent the OR gate. They are equivalent to 

```
po0 <-> pi00 OR pi01
```

The clause 

2 0

effectively tells the solver to look for an assignment that satisfies `po0`.

The remaining clause 

```
-5 0
```

creates a variable whose value is zero, which can be used instead of the constant zero if it was present in the input  `.bench` file. This could have been optimized away as constants are not used in the input `.bench` file.