berkeley-abc/abc #60: [Feature] How to write the output file in topological order?

MahdiNazemi, 2019-12-01:
I have a combinational circuit that is optimized in ABC. When writing the optimized design using the `write_verilog` command, I get `assign` statements that are not topologically sorted and therefore, make understanding of the code difficult.

I have read in an online powerpoint presentation of ABC that the internal AIG keeps the nodes in topological order. Is it possible to write them in a file using the same order? How about for a network that is not AIG (e.g. `multi` is called before writing the file).

alanminko, 2019-12-01:
AIG nodes are always written in a topological order.  It is only the 
primary output drivers that are written as assign-statement may be in a 
different order (which is in fact the order in which the primary outputs 
are listed in the input file, and not in the topological order).

Please feel free to try command "&write -p <file.v>" (or "&get; &write 
-p <file.v>" if you are using the main-space).

MahdiNazemi, 2019-12-02:
You're right. Writing an AIG to file will print the nodes in topological order. However, when I run `multi` on the same design, the order of `assign` statements is not in topological order and some statements consume the output of nodes that come after them.

jimouris, 2023-05-16:
Hi @MahdiNazemi, @alanminko, did you figure out a solution to this? I'm facing the same problem with the `assign` statements not being topologically sorted.