berkeley-abc/abc #70: LUT2 mapping

killruana, 2020-03-09:
I have tried to map an application to LUT of size 2. But the resulting netlist contains LUT 2 and 3.
I use abc through yosys (I don't know how to use abc alone).

My RTL:
```
module Adder8(z, b, a);
    output [7:0] z;
    input [7:0] a, b;

    assign z = a + b;
endmodule
```

My synthesis script:
```
proc; opt; fsm; opt; memory; opt
techmap; opt
abc -lut 2 -dress; opt
```

The netlist:
```
/* Generated by Yosys 0.9+1706 (git sha1 8b074cc4, gcc 9.2.1+20200130-2 -march=native -O3 -fno-plt -fPIC -Os) */

(* top =  1  *)
(* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:3.1-8.10" *)
module Adder8(z, b, a);
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:0|<techmap.v>:295.27-295.69|<techmap.v>:238.21-238.22" *)
  wire _00_;
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:0|<techmap.v>:295.27-295.69|<techmap.v>:238.21-238.22" *)
  wire _01_;
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:0|<techmap.v>:295.27-295.69|<techmap.v>:238.21-238.22" *)
  wire _02_;
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:0|<techmap.v>:295.27-295.69|<techmap.v>:238.21-238.22" *)
  wire _03_;
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:0|<techmap.v>:295.27-295.69|<techmap.v>:238.21-238.22" *)
  wire _04_;
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:0|<techmap.v>:295.27-295.69|<techmap.v>:238.21-238.22" *)
  wire _05_;
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:0|<techmap.v>:295.27-295.69|<techmap.v>:238.21-238.22" *)
  wire _06_;
  wire _07_;
  wire _08_;
  wire _09_;
  wire _10_;
  wire _11_;
  wire _12_;
  wire _13_;
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:5.17-5.18" *)
  input [7:0] a;
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:5.20-5.21" *)
  input [7:0] b;
  (* src = "/home/killruana/projects/kfpga_workspace/testsuite/apps/adder8/rtl/Adder8.v:4.18-4.19" *)
  output [7:0] z;
  assign z[1] = 4'h6 >> { _07_, _00_ };
  assign _00_ = 4'h8 >> { b[0], a[0] };
  assign _07_ = 4'h6 >> { b[1], a[1] };
  assign z[2] = 4'h9 >> { _08_, _01_ };
  assign _08_ = 4'h6 >> { b[2], a[2] };
  assign z[3] = 4'h9 >> { _09_, _02_ };
  assign _09_ = 4'h6 >> { b[3], a[3] };
  assign z[4] = 4'h9 >> { _10_, _03_ };
  assign _10_ = 4'h6 >> { b[4], a[4] };
  assign z[5] = 4'h9 >> { _11_, _04_ };
  assign _11_ = 4'h6 >> { b[5], a[5] };
  assign z[6] = 4'h9 >> { _12_, _05_ };
  assign _12_ = 4'h6 >> { b[6], a[6] };
  assign _13_ = 4'h9 >> { b[7], a[7] };
  assign z[0] = 4'h6 >> { b[0], a[0] };
  assign _01_ = 8'h17 >> { _00_, b[1], a[1] };
  assign _02_ = 8'h71 >> { _01_, b[2], a[2] };
  assign _03_ = 8'h71 >> { _02_, b[3], a[3] };
  assign _04_ = 8'h71 >> { _03_, b[4], a[4] };
  assign _05_ = 8'h71 >> { _04_, b[5], a[5] };
  assign _06_ = 8'h71 >> { _05_, b[6], a[6] };
  assign z[7] = 4'h6 >> { _13_, _06_ };
endmodule
```
The abc script used by yosys for the `abc -lut 2` command is `strash; ifraig; scorr; dc2; dretime; retime {D}; strash; dch -f; if; mfs2; lutpack {S}`. Someone from the [Yosys subreddit](https://www.reddit.com/r/yosys/comments/fe4zod/map_to_lut2/fjqgmw5/) pointed out that the culprit is probably the `lutpack` command and recommended me to notify the upstream, so here I am :)

yupferris, 2022-12-07:
I've also encountered this issue.

yupferris, 2022-12-07:
This is probably user error actually. If just LUT mapping is desired, the following works as expected: `read in.v; if -K 2; write out.v`

resulting `out.v`:
```verilog
// Benchmark "in" written by ABC on Wed Dec  7 11:21:56 2022

module in ( 
    a, b, c,
    d  );
  input  a, b, c;
  output d;
  wire new_n7_;
  assign d = new_n7_ | c;
  assign new_n7_ = a & b;
endmodule
```

If trying to do this from yosys, a similar script should be provided.