berkeley-abc/abc #72: src/base/io/ioReadAiger.c:435: Io_ReadAiger: Assertion failed

muchang, 2020-03-24:
Hi,
For this Aiger model (both aig and aag included in the zip file):
[assertion.zip](https://github.com/berkeley-abc/abc/files/4373272/assertion.zip)
ABC throws out an assertion failed with pdr:
```
$ abc/abc -c "read assertion.aig;pdr -g;write_cex -a out.cex"
ABC command line: "read assertion.aig;pdr -g;write_cex -a out.cex".

Warning: The last 2 outputs are interpreted as constraints.
abc: src/base/io/ioReadAiger.c:435: Io_ReadAiger: Assertion `Init == Abc_Var2Lit(1+Abc_NtkPiNum(pNtkNew)+i, 0)' failed.
Aborted
```
Is it the expected behavior? Thanks!

OS: Ubuntu 18.04
Commit: c7bc6b6

sterin, 2020-03-25:
The AIGER file is not well-formed. 

It's most easily seen in the .aag file. The following line represents a latch:

    74 1926 75

The first number is the literal of the latch, the second is the literal for the next state, and the third is the initial value.

According to the AIGER [spec](http://fmv.jku.at/hwmcc11/beyond1.pdf), the initial value should be either 0, 1, or the literal of the latch, in this case 74.

closed 2020-03-25 by sterin

muchang, 2020-03-26:
I see, thank you for the explanation!