berkeley-abc/abc #73: Verification success on unsafe model?

muchang, 2020-03-26:
Hi,
For this .aig model:
```
aig 1 0 1 1 0 0 1
0
0
3
```
(corresponding .aag model is:
```
aag 1 0 1 1 0 0 1
2 0
0
3
```
)
abc thinks the model is safe:
```
$ ./abc/abc -c "read reduced.aig;pdr -g;write_cex -a out.cex"
ABC command line: "read reduced.aig;pdr -g;write_cex -a out.cex".

Warning: The last output is interpreted as a constraint.
Warning: The new network has no primary inputs. It is recommended
to add a dummy PI to make sure all commands work correctly.
Invariant F[1] : 1 clauses with 1 flops (out of 1) (cex = 0, ave = 1.00)
Verification of invariant with 1 clauses was successful.  Time =     0.00 sec
Property proved.  Time =     0.02 sec
Counter-example is not available.
$ cat reduced.aig
aig 1 0 1 1 0 0 1
0
0
3
``` 
However, there is a valid counter-example according to the aigsim tool:
```
$ ./aiger-1.9.4/aigsim -c reduced.aig out
[aigsim] WARNING no bad state properties found using outputs instead
$ cat out
1
b0
0

.
```
Is it the expected behavior or the bug?

OS: Ubuntu 18.04
Commit: a277d45

sterin, 2020-03-27:
A few comments:

First, you need to check the exit status of `aigsim`. You will find out that it is 1, which means that the cex is not valid.

The reason is that the first output is the literal 0, which corresponds to constant 0. In AIGER, a valid counter example (cex) must have at least one state (usually the last) where the output is 1. The best way to think about it is that the outputs represent whether the state is bad, so the cex must contain a bad state.

Second, ABC was written in part before AIGER 1.9, so there are some quirks when working with AIGER 1.9 files. There is a recommended sequence of commands that should to be run before and after verification.

Start by reading the AIGER file

    read reduced.aig

Convert non-deterministically initialized latches to constant initialized latches with `undc`, then convert non-zero initialized latches to zero-initialized latches with `zero`.

    logic
    undc
    zero
    strash

Fold the constraints into the properties

    fold

Now it is safe to use the various verification commands. 

    pdr

The problem is that the counter example is in terms of the transformed AIGER. To transform it back, you need to read the original AIGER, and use `udc -c` to transform back the cex. Reading a new AIG removed the cex, so it needs to be saved then load.

    cexsave
    read reduced.aig
    cexload
    undc -c

And then finally write the cex

    write_cex -a cex

These commands are very fast, even on large AIGs.

closed 2020-03-27 by sterin

muchang, 2020-03-27:
Thanks a lot for your detailed explanation. 
I will try to use the recommended way in the future. 

Just curious, whether the quirks you mentioned are something like:
```
$ ./abc/abc -c "read reduced.aig;pdr;write_cex -a out.cex"ABC command line: "read reduced.aig;pdr;write_cex -a out.cex".

Warning: The new network has no primary inputs. It is recommended
to add a dummy PI to make sure all commands work correctly.
Warning: 1 registers in this network have don't-care init values.
The don't-care are assumed to be 0. The result may not verify.
Use command "print_latch" to see the init values of registers.
Use command "zero" to convert or "init" to change the values.
Invariant F[1] : 1 clauses with 1 flops (out of 1) (cex = 0, ave = 1.00)
Verification of invariant with 1 clauses was successful.  Time =     0.00 sec
Property proved.  Time =     0.02 sec
Counter-example is not available.
$ ./abc/abc -c "read reduced.aig;logic;undc;strash;zero;pdr;write_cex -a out.cex"
ABC command line: "read reduced.aig;logic;undc;strash;zero;pdr;write_cex -a out.cex".

Warning: The new network has no primary inputs. It is recommended
to add a dummy PI to make sure all commands work correctly.
The number of converted latches with DC values = 1.
Output 0 of miter "reduced" was asserted in frame 0.  Time =     0.01 sec
$ cat reduced.aig 
aig 1 0 1 0 0 1
0 2
2
$ cat reduced.aag 
aag 1 0 1 0 0 1
2 0 2
2
```
Here, abc with `pdr;write_cex -a out.cex` produces success, while abc with the recommended commands sequence `logic;undc;strash;zero;` produces asserted. 

So the second result is correct, right?

sterin, 2020-03-27:
Yes.

Many ABC commands assume that latches are zero-initialized. 

So your `reduced.aig` model is just a single latch that is initialized to zero and its next state is zero. Therefore the output stays zero forever and therefore considered safe.

Similarly, if you have latches initialized to one, they are still treated as if they were initialized to zero.

Finally, constraints will be treated as regular outputs without `fold`.

muchang, 2020-03-27:
Thanks for providing the information, but it is still a little bit unclear for me. 

As you said,  if ABC considers the latches are zero-initialized, especially with `undc;zero;` commands, the `reduced.aig` model should be safe. 
However, `./abc/abc -c "read reduced.aig;logic;undc;strash;zero;pdr;write_cex -a out.cex` results in unsafe according to my above comment, which confilts to our assumption.

Do I misunderstand something?

sterin, 2020-03-27:
Let me try to make it a bit clearer.

ABC (mostly) assumes all latches are zero initialized, and treats all outputs the same way, regardless of whether they're outputs, bad-state outputs, or constraints.

In other words, ABC would treat the original reduced.aig as if the latch was zero-initialized, and pdr would claim reduced.aig is safe. This would be a very bad outcome, as reduced.aig has a cex.

The purpose of the command sequence `undc;zero;fold` is to transform an AIG that potentially has non-zero initialized latches and/or constraints into an **equivalent** AIG with only zero-intialized latches and property outputs. This equivalent AIG would be handled correctly, and pdr would return the right cex.

muchang, 2020-03-28:
I see, the explanation is super clear! :) Thanks!
I will use command sequence `undc;zero;fold` on AIGER 1.9 files in the future to make ABC work correctly.

sirandreww, 2024-03-12:
> A few comments:
>
> First, you need to check the exit status of `aigsim`. You will find out that it is 1, which means that the cex is not valid.
>
> The reason is that the first output is the literal 0, [...]

Shouldn't `zero` be called after `strash` as per https://github.com/berkeley-abc/abc/issues/40#issuecomment-493717330

sterin, 2024-03-20:
> Shouldn't zero be called after strash as per https://github.com/berkeley-abc/abc/issues/40#issuecomment-493717330

Yes, `strash` should be added before `zero`. The `zero` command reports an error if this is not done. I'll update the original answer.