berkeley-abc/abc #75: bmc2 Segmentation fault

muchang, 2020-04-09:
Hi,
For this model:
[Segfault.zip](https://github.com/berkeley-abc/abc/files/4454983/Segfault.zip)

ABC throws out a segmentation fault with bmc2 command:
```
$ abc/abc -c "read segfault.aig;logic;undc;strash;zero;fold;bmc2;" 
ABC command line: "read segfault.aig;logic;undc;strash;zero;fold;bmc2;".

Warning: The last 10 outputs are interpreted as constraints.
Segmentation fault
```
While pdr command works well on it:
```
$ abc/abc -c "read segfault.aig;logic;undc;strash;zero;fold;pdr;" 
ABC command line: "read segfault.aig;logic;undc;strash;zero;fold;pdr;".

Warning: The last 10 outputs are interpreted as constraints.
Invariant F[1] : 1 clauses with 1 flops (out of 3183) (cex = 0, ave = 24.00)
Verification of invariant with 1 clauses was successful.  Time =     0.00 sec
Property proved.  Time =     0.06 sec
```

OS: Ubuntu 18.04
Commit: dab7168

alanminko, 2020-04-09:
Thank you for the report.  After a quick look, it seems that "bmc2" is 
running out of memory.  It would be hard to address this, without 
redesigning the data-structure.  On the other hand, the miter appears to 
be UNSAT and it is quickly proved by "pdr".

muchang, 2020-04-10:
Thanks, Alan! 
Besides "bmc2", "int", and "pdr", what other commands (solvers) I could use for solving an AIG model?  
Are there any documents or tutorials for it?

alanminko, 2020-04-10:
This paper may be the closest we have to a tutorial on verification 
commands in ABC:

https://people.eecs.berkeley.edu/~alanmi/publications/2010/cav10_abc.pdf

muchang, 2020-04-11:
Thanks for your guidence, Alan! I will check it out.

muchang, 2020-04-18:
This segmentation fault seems to be fixed. Should we close this issue?