berkeley-abc/abc #77: bmc2 Segmentation fault (core dump)

muchang, 2020-04-18:
Hi,
For this model:
[segfault.zip](https://github.com/berkeley-abc/abc/files/4496496/segfault.zip)

ABC throws out a segmentation fault with bmc2 command on both original model and reduced model, while pdr works well:
```
$ abc/abc -c "read segfault.aig;logic;undc;strash;zero;fold;bmc2;"
ABC command line: "read segfault.aig;logic;undc;strash;zero;fold;bmc2;".

Warning: The last 2 outputs are interpreted as constraints.
Segmentation fault (core dumped)

$ abc/abc -c "read reduced.aig;logic;undc;strash;zero;fold;bmc2;"
ABC command line: "read reduced.aig;logic;undc;strash;zero;fold;bmc2;".

Warning: The network has no constraints.
Segmentation fault (core dumped)

$ abc/abc -c "read segfault.aig;logic;undc;strash;zero;fold;pdr;"
ABC command line: "read segfault.aig;logic;undc;strash;zero;fold;pdr;".

Warning: The last 2 outputs are interpreted as constraints.
Invariant F[5] : 9 clauses with 10 flops (out of 42) (cex = 0, ave = 21.93)
Verification of invariant with 9 clauses was successful.  Time =     0.00 sec
Property proved.  Time =     0.02 sec

$ abc/abc -c "read reduced.aig;logic;undc;strash;zero;fold;pdr;"
ABC command line: "read reduced.aig;logic;undc;strash;zero;fold;pdr;".

Warning: The network has no constraints.
Output 0 of miter "reduced" was asserted in frame 16384.  Time =    21.19 sec

$ cat reduced.aag 
aag 66 1 16 0 49 1
2
4 63
6 66
8 70
10 74
12 78
14 82
16 86
18 90
20 94
22 98
24 102
26 106
28 110
30 114
32 33
34 126
132
36 32 30
38 36 28
40 38 26
42 40 24
44 42 22
46 44 20
48 46 18
50 48 16
52 50 14
54 52 12
56 54 10
58 56 8
60 58 6
62 61 5
64 59 7
66 65 61
68 57 9
70 69 59
72 55 11
74 73 57
76 53 13
78 77 55
80 51 15
82 81 53
84 49 17
86 85 51
88 47 19
90 89 49
92 45 21
94 93 47
96 43 23
98 97 45
100 41 25
102 101 43
104 39 27
106 105 41
108 37 29
110 109 39
112 33 31
114 113 37
116 34 2
118 35 3
120 119 117
122 120 2
124 121 3
126 125 123
128 35 2
130 128 3
132 131 4
```
It should be different from https://github.com/berkeley-abc/abc/issues/75, since https://github.com/berkeley-abc/abc/issues/75 seems to be fixed and does not have (core dump).

OS: Ubuntu 18.04
Commit: 94a2cff

alanminko, 2020-04-18:
There are two reasons why "bmc2" may crash: running out of memory and 
reaching the limit on the number of recursive calls.  It is recommended 
to use "bmc3", which is overall better designed when it comes to both of 
these situations.

muchang, 2020-04-19:
I see, thanks, Alan! I will use "bmc3" instead.