berkeley-abc/abc #78: Word Level Identification

fvrmatteo, 2020-04-19:
Hi,

I'm really new to this field and I've been reading some papers about the topic of "word level components" identification like the following:

- **Structural Reverse Engineering of Arithmetic Circuits**
- **WordRev: Finding Word-Level Structures in a Sea of Bit-Level Gates**
- **Reverse Engineering Digital Circuits Using Functional Analysis**
- **Reverse Engineering Circuits Using Behavioral Pattern Mining**
- **Automatic Word-level Abstraction of Datapath**
- **Rewriting Environment for Arithmetic Circuit Verification** (polynomials as output)

I noticed that some of the exposed ideas have been implemented in ABC, but, while diving in the **help -d** command, I was wondering if there is a suggested way to achieve the following (with ABC commands):

- identify candidate word cuts via _shape hashing_ and/or _bit-slice aggregation_;
- automatic word-level abstraction (of AND, OR, XOR, ADD, MUL n-bit components).

In my experiments I'm starting with a word-level representation written in Verilog, I'm converting it to BLIF with Yosys and then I would like to ideally get back to the word-level representation (or to having a contained amount of candidate word cuts to apply some kind of synthesis of the sub-circuit). In the examples I'm also using hardcoded constants that are propagated in the circuit (hindering the "shape" of the components):

```
module example (a, b, c, o);
//-------------Input Ports Declarations----------------------------
input [3:0]a;
input [3:0]b;
input [3:0]c;
//-------------Output Ports Declarations---------------------------
output [3:0]o;
//-------------Wires Declarations----------------------------------
wire [3:0]w1;
wire [3:0]w2;
wire [3:0]w3;
wire [3:0]w4;
//-------------Logic-----------------------------------------------
assign w1 = (a * b);
assign w2 = (w1 | c);
assign w3 = (w2 + 8'h3);
assign w4 = (w3 - w1);
assign o = w4;
endmodule
```

It's really likely that I'm missing knowledge of the current state-of-the-art on the topic, so I won't mind reading more papers that promise an open source (or well documented) solution. It's my understanding that the polynomial-based rewriting techniques to obtain the circuit signature on the inputs is the current best approach for the verification problem, but I didn't find proper research on using polynomials for word-level abstraction (except for the multiplier P(x) identification in **Reverse Engineering of Irreducible Polynomials in GF(2^m) Arithmetic**).

alanminko, 2020-04-19:
One of our recent students has worked on reverse engineering, among 
other things, and came up with a method to detect "transparent logic", 
which is somewhat related to "bit-slice aggregation". Please have a look 
at Chapter 3 in this thesis:

https://www2.eecs.berkeley.edu/Pubs/TechRpts/2017/EECS-2017-122.pdf

Regarding the use of polynomials for arithmetic circuit verification, 
this method is effective for artificially generated multipliers.  
However, for industrial benchmarks, with possible truncation (not using 
some output bits), constant propagation on the
input side, or minor synthesis, this method does not work.  This was our 
conclusion after working on this project several years ago.

There were several papers published on this topic, but the code in ABC 
has not reached quality sufficient to recommend its public use.

https://msoeken.github.io/papers/2015_fmcad.pdf

https://people.eecs.berkeley.edu/~alanmi/publications/2017/tech17_arith.pdf

fvrmatteo, 2020-04-21:
Thanks for pointing out that PhD thesis and the **Simulation Graphs for Reverse Engineering** paper. The additional "transparent logic" approach is interesting and somehow complementary to the other mentioned techniques.

I'll keep an eye on the new additions to ABC to see if in the futures some of the word-level identification techniques will be made public.

closed 2020-04-21 by fvrmatteo