berkeley-abc/abc #81: --best chooses abc and contrib/get-abc fails

pmatos, 2020-05-11:
```
$ contrib/get-abc
Using CC=gcc
Using CXX=g++
Using LD=g++
Compiling with CUDD
Found GCC_VERSION 9.3.0
Found GCC_MAJOR>=4
Using CFLAGS=-Wall -Wno-unused-function -Wno-write-strings -Wno-sign-compare -arch x86_64 -DABC_USE_CUDD=1
`` Generating dependency: /src/bdd/llb/llb4Sweep.c
`` Generating dependency: /src/bdd/llb/llb4Nonlin.c
`` Generating dependency: /src/bdd/llb/llb4Image.c
`` Generating dependency: /src/bdd/llb/llb4Cex.c
`` Generating dependency: /src/bdd/llb/llb3Image.c
`` Generating dependency: /src/bdd/llb/llb3Nonlin.c
`` Generating dependency: /src/bdd/llb/llb2Image.c
`` Generating dependency: /src/bdd/llb/llb2Flow.c
`` Generating dependency: /src/bdd/llb/llb2Dump.c
`` Generating dependency: /src/bdd/llb/llb2Driver.c
`` Generating dependency: /src/bdd/llb/llb2Core.c
`` Generating dependency: /src/bdd/llb/llb2Bad.c
`` Generating dependency: /src/bdd/llb/llb1Sched.c
`` Generating dependency: /src/bdd/llb/llb1Reach.c
`` Generating dependency: /src/bdd/llb/llb1Pivot.c
`` Generating dependency: /src/bdd/llb/llb1Matrix.c
gcc: error: x86_64: No such file or directory
gcc: error: x86_64: No such file or directory
gcc: error: x86_64: No such file or directory
gcc: error: x86_64: No such file or directory
gcc: error: unrecognized command line option ‘-arch’
...
```

alanminko, 2020-05-11:
Please make sure you have an up-to-date version of GCC.  Another idea: 
try to remove that flag and see what happens.

pmatos, 2020-05-11:
@alanminko my gcc is 9.3.0
there's no arch option. Maybe you meant `-march`?

sterin, 2020-05-11:
@pmatos The ABC Makefile does not set the `-arch` flag.  This has to come from outside, probably through the `ARCHFLAGS` environment variable.

This is a pure guess: `-arch` is a clang flag, so I assume you are working on macOS, and you might have an environment variable `ARCHFLAGS` set to `-arch x86_64`. 

Please unset the variable and try again.

pmatos, 2020-05-11:
Interesting - I am actually on arch linux, so I am confused as to where those flags are coming from. I will take a look. Thanks for your time.

pmatos, 2020-05-11:
ah - wait, this is being built by cvc4 so cvc4 must be adding these flags. will confirm.

pmatos, 2020-05-11:
OK - sorry, this is an issue with cvc4.

closed 2020-05-11 by pmatos