berkeley-abc/abc #94: bmc3 Segmentation fault

muchang, 2020-08-03:
Hi,
For this model:
[reduced.zip](https://github.com/berkeley-abc/abc/files/5017398/reduced.zip)

ABC throws out a segmentation fault with bmc3 command, while pdr and bmc2 work well. It seems due to the running out of the memory. It may take more time to reproduce on the machine with larger memory:
```
$ time ../abc/abc -c "read reduced.aig;logic;undc;strash;zero;fold;bmc3"
ABC command line: "read reduced.aig;logic;undc;strash;zero;fold;bmc3".

Warning: The last 127 outputs are interpreted as constraints.
Warning: The new network has no primary inputs. It is recommended
to add a dummy PI to make sure all commands work correctly.
Segmentation fault

real    0m46.831s
user    0m12.931s
sys     0m10.151s
$ time ../abc/abc -c "read reduced.aig;logic;undc;strash;zero;fold;pdr"
ABC command line: "read reduced.aig;logic;undc;strash;zero;fold;pdr".

Warning: The last 127 outputs are interpreted as constraints.
Warning: The new network has no primary inputs. It is recommended
to add a dummy PI to make sure all commands work correctly.
Invariant F[1] : 1 clauses with 1 flops (out of 23) (cex = 0, ave = 23.00)
Verification of invariant with 1 clauses was successful.  Time =     0.00 sec
Property proved.  Time =     0.02 sec

real    0m0.140s
user    0m0.052s
sys     0m0.010s
$ time ../abc/abc -c "read reduced.aig;logic;undc;strash;zero;fold;bmc2"
ABC command line: "read reduced.aig;logic;undc;strash;zero;fold;bmc2".

Warning: The last 127 outputs are interpreted as constraints.
Warning: The new network has no primary inputs. It is recommended
to add a dummy PI to make sure all commands work correctly.
^C
real    5m34.015s
user    2m44.688s
sys     0m0.178s

$ cat reduced.aag 
aag 41 0 22 0 19 1 127
2 0
4 0
6 0
8 0
10 0
12 0
14 0
16 0
18 0
20 0
22 0
24 0
26 0
28 0
30 0
32 0
34 0
36 36
38 38
40 40
42 42
44 44
1
52
66
1
1
82
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
46 38 36
48 46 40
50 48 42
52 50 44
54 5 3
56 54 7
58 56 9
60 58 11
62 60 13
64 62 15
66 64 17
68 21 19
70 68 23
72 70 25
74 72 27
76 74 29
78 76 31
80 78 33
82 80 35
```

OS: Ubuntu 18.04
Machine memory: 2G
Commit: 5c8ee4a

alanminko, 2020-08-03:
In this case, "bmc3" is indeed running out of memory. Please consider 
using a bound on the number of timeframes to be unfolded, for example:

> abc 01> bmc3 -F 1000
> No output asserted in 1000 frames. Resource limit reached (conf limit 
> 0). Time =     0.07 sec
> abc 02> bmc3 -F 10000
> No output asserted in 10000 frames. Resource limit reached (conf limit 
> 0). Time =     0.26 sec
> abc 03> bmc3 -F 100000
> No output asserted in 100000 frames. Resource limit reached (conf 
> limit 0). Time =     2.48 sec

muchang, 2020-08-04:
Thank you for your suggestion, although it may not be adaptable for the people who would like to limit the running time rather than the timeframes. In this case, it would be fine. 
Here is the further reduced model which would be helpful to you:
[reduced_futher.zip](https://github.com/berkeley-abc/abc/files/5019812/reduced_futher.zip)
```
aag 41 0 22 0 19 1 3
2 0
4 0
6 0
8 0
10 0
12 0
14 0
16 0
18 0
20 0
22 0
24 0
26 0
28 0
30 0
32 0
34 0
36 36
38 38
40 40
42 42
44 44
1
52
66
82
46 38 36
48 46 40
50 48 42
52 50 44
54 5 3
56 54 7
58 56 9
60 58 11
62 60 13
64 62 15
66 64 17
68 21 19
70 68 23
72 70 25
74 72 27
76 74 29
78 76 31
80 78 33
82 80 35
```