berkeley-abc/abc #96: read_library latch.genlib

Flians, 2020-08-27:
Hi,
Why does the program crash when It reads a .genlib file containing LATCH?

```
C:\Users\Administrator\Desktop\ABC\ABC\reference>abc80930.exe
UC Berkeley, ABC 1.01 (compiled Sep 30 2018 23:38:01)
================== Command history ==================
source abc.rc
=====================================================
Loading resource file "abc.rc".
abc 01> read_library rsfq.genlib
Skipping latch "jdff"...
Skipping latch "jnot"...
Skipping latch "jand"...
Skipping latch "jor"...
Skipping latch "jxor"...
Entered genlib library with 5 gates from file "rsfq.genlib".

C:\Users\Administrator\Desktop\ABC\ABC\reference>
```

liy997, 2021-01-22:
May I ask where can I find the .genlib file？

alanminko, 2021-01-22:
SIS distribution has some genlib files in it.

https://ptolemy.berkeley.edu/projects/embedded/Alumni/pchong/sis.html

There is also SIS manual which shows how to create them.

https://www.researchgate.net/publication/2624520_SIS_A_System_for_Sequential_Circuit_Synthesis

Flians, 2021-01-25:
> May I ask where can I find the .genlib file？

You can refer to the "genlib-format.pdf" to write the .genlib file
[genlib-format.pdf](https://github.com/berkeley-abc/abc/files/5863778/genlib-format.pdf)

closed 2021-01-25 by Flians