berkeley-abc/abc #98: File size limit?

jbroot, 2020-09-24:
When reading in a Bench file with size ~4.4 GB I get this error:
> file.bench: Wrong input file format.
> Reading network from file has failed.
The bench file is generated by program that has always produced well-formed Bench files in the past, and the smaller Bench files produced are accepted by ABC.
I tried ABC from my build on June 12, 2020, and again with the most recent build (September 23, 2020).

alanminko, 2020-09-24:
Most likely, there is an unguarded memory overlfow at 4GB file size 
limit. I would highly recommend migrating to using AIGER format 
http://fmv.jku.at/aiger/ instead of BENCH.  This way you will be able to 
work with up to 2^29=536M objects.  While in a 4.4GB BENCH file, you 
like only have ~10M objects.

I am currently developing new features to deal with large netlists in 
ABC and would be happy to know more about a realistic use-case of this 
type.  If possible, please tell me more about the project. Why do you 
need large flat netlists?  Is there are a way to represent them in a 
more compact hierarchical form?

jbroot, 2020-09-24:
For my thesis, I am converting neural networks to a layered sum-of-products (remembered as a truth table with don't care variables) then to AIGs. Some neural networks yield very simple AIGs (after simplification) despite the network's high complexity--imagine a 100X100X100X1 neural network that always predicts true. I am also using mockturtle's AIG C++17 implementation, the intersection of AIG file types mockturle can write and file types ABC can read are {blif, verilog, bench}. 

On the neural networks that represent inherently complex relations, the simplified AIG's node per level increases faster than the number of levels. For example, a particular 14X14X14X1 neural network yields 158,665 nodes and 52 levels (~3051 nodes/level), and a pruned version of that network (10X10X10X1) yields 12,696 nodes and 40 levels (~317 nodes/level). However, we cannot know how simple a large AIG can be without simplifying it. 

I am open to ideas that require change such as using different software to compile file formats or using a structure other than AIGs.

alanminko, 2020-09-25:
Thank you for the explanation.

I suspect you are creating one circuit per layer, not one circuit for 
the whole neural network?  In this case, it would be better to start 
with a list of minterms in the PLA form and then apply command "fxch" in 
ABC to derive an AIG. 
https://people.eecs.berkeley.edu/~alanmi/publications/2017/aspdac17_fxch.pdf 
The benchmarks are attached, in case you would like to see how they look 
in the PLA format.

What quantization do you use? (definitely the fewer bits the better)

jbroot, 2020-09-26:
I create an AIG for each node in a layer using the sum-of-products. Each of those AIGs is aggregated into one by parallel. Then each layer's AIG becomes part of the entire neural network's AIG by combining the layers sequentially--the outputs of layer one are the inputs for layer two.
FXCH looks far better suited for my needs than what I've been doing. However, I noticed in https://people.eecs.berkeley.edu/~alanmi/abc/abc.htm it says: "[o]nly binary PLAs are currently supported; the don’t-care cubes are currently ignored." Will this require me to expand "1--1" to "1001, 1011, 1101, 1111"? Also, does ABC support agglomerating truth-tables (which represent nodes) to layers (in parallel) and layers to the network (sequential) easily?

Regarding quanization, I cast all input values into {0,1}. This works fine for some inputs, however other inputs have a non-monotonic relationship with the output and too much information is lost with the {0,1} cast. In the latter case, single inputs can be read as two or more inputs which can effectively maintain information. If I understand "Making Logic Learnable With Neural Networks" by Brudermueller et al. correctly, those mad lads trained a decision tree on each bit of each input. While this method is very inefficient, it would certainly maintain all information.

alanminko, 2020-09-26:
You are right that ABC does not take SOPs with don't-cares.  This is why 
it would sense to take an incompletely specified function as a set of 
onset/offset minterms and minimize it using Espresso, which will create 
SOPs similar to those included in the benchmarks, which I sent before, 
and these can be fed into ABC for multi-level synthesis.

I understand the usefulness of aggregating layers in parallel, but I 
wonder if it is wise to combine layers sequentially.  It may be better 
to keep each layer in a separate AIG file/manager, to avoid the 
explosion of the AIG size, and maintain the modularity of the design.

I am familiar with the work of Brudermueller. Indeed they minimized each 
output bit.  This work takes yet another approach to the same problem: 
https://arxiv.org/abs/2004.03021

jbroot, 2020-09-27:
That paper reminds me of Chatterjee's "Learning and Memorization" http://proceedings.mlr.press/v80/chatterjee18a/chatterjee18a.pdf . Both researchers even called the product "LogicNet", and I cannot see either paper reference the other. Basically, Chatterjee found that truth tables (memorization) can generalize when arranged in a network.

My thesis was driven with the intention to elucidate neural networks and human comprehension of the end product has been a large focus. My understanding that comprehension must be modular has been reinforced during this research and the layer separation could be beneficial to human comprehension. However, another factor is conciseness. I combined all layers so ABC could simplify the AIG and reduce the total amount of nodes--producing a more concise network (unfortunately some functions cannot be abstracted to be accurate and understandable). Additionally, I can see how layer separation would be useful for debugging or future work. I plan on finishing my thesis this fall semester, then using the lessons learned to further improve my work for publication before or during my PhD (starting fall 2021 and location TBD--I am just starting applications).

alanminko, 2020-09-29:
Thank you for explaining your viewpoint.  It is true that human 
comprehension is desirable but hard to achieve in a larger model, except 
perhaps in the last layer.  I am well familiar with Chatterjee's work, 
reimplemented it and found difficult to improve on / move forward.  
Please feel free to share your thesis with me when it reaches the stage 
of being readable.

jbroot, 2020-10-01:
Of course! I would love to get some feedback and ideas on my thesis. Will "alanmi at EECS dot Berkeley dot edu" be ok to email my thesis when it's readable?