berkeley-abc/abc #112: Running abc commands by sourcing a bash scripts

rafidmuttaki, 2020-12-05:
Hi,

I am trying to do a specific number of iteration where in each iteration, abc tool needs to read a verilog file. However, I can see that the tool is invoked and stays in the same line and does not execute the subsequent commands. Can someone please help. It's kind of urgent. Thank you!

Example: exa.sh has following lines:
./abc
read_library cadence.genlib
read -m verilog.v

alanminko, 2020-12-05:
Please note that ABC may read the original IWLS'05 benchmarks in Verilog 
but not much else. The build it parser is extremely weak and buggy.

mithro, 2020-12-05:
Random comment which I'm not sure makes total sense, maybe it would be possible to use [Surelog](https://github.com/alainmarcel/Surelog) and [UHDM](https://github.com/alainmarcel/UHDM) as a replacement parser to read Verilog into ABC? There is already a bunch of work being done to integrate Surelog with UHDM into Yosys and Verilator at https://github.com/alainmarcel/uhdm-integration

rrlcs, 2022-01-04:
> Hi,
>
> I am trying to do a specific number of iteration where in each iteration, abc tool needs to read a verilog file. However, I can see that the tool is invoked and stays in the same line and does [...]

Were you able to solve this?

rrlcs, 2022-01-04:
One can use this command

./abc -c "read_library cadence.genlib; read -m verilog.v"

alanminko, 2022-01-04:
Please try this: ./abc -c "read_library cadence.genlib; read -m verilog.v"