berkeley-abc/abc #114: Segmentation fault at ./src/misc/vec/vecInt.h:460

muchang, 2020-12-17:
Hello, here is an AIGER input that triggers a segmentation fault when using tempor. 
[trigger.aig.zip](https://github.com/berkeley-abc/abc/files/5707515/trigger.aig.zip)

Here is the trace:
```
$ ./abc -c "read trigger.aig;tempor"  
ABC command line: "read trigger.aig;tempor".

Using computed frame number (3041).
Segmentation fault
$ gdb ./abc
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./abc...done.
(gdb) run -c "read trigger.aig;tempor"
Starting program: /home/abc/abc -c "read trigger.aig;tempor"
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
ABC command line: "read trigger.aig;tempor".

Using computed frame number (3041).

Program received signal SIGSEGV, Segmentation fault.
0x0000555555c02f60 in Vec_IntWriteEntry (Entry=0, i=0, p=0x55555a1e7240)
    at ./src/misc/vec/vecInt.h:460
460	    p->pArray[i] = Entry;
(gdb) 
```

Operating System: Ubuntu 18.04
Commit ID: 73dcdab

alanminko, 2020-12-17:
After a quick look, this appears to be an out-of-memory.  There is 
nothing to do about it, other than getting a bigger computer or not 
using this command.

muchang, 2020-12-18:
Thanks. We will try to use a larger memory. 

BTW, is `tempor` a bmc algorithm?
If so, when `tempor` says:
```
No output failed in n frames. 
Reached limit on the number of timeframes (n).
``` 
Does it mean that this network is safe?

alanminko, 2020-12-20:
Command "tempor" performs an internal BMC run. If the property fails, 
the transformation is not applied.  If the BMC run passes, it means that 
the property holds for at least n frame, and temporal decomposition can 
be applied as described in this paper 
https://www.research.ibm.com/haifa/projects/verification/SixthSense/papers/fmcad09-ffd.pdf

muchang, 2020-12-22:
I see, thank you! It helps.