berkeley-abc/abc #115: Assertion `nInts + 3 < (1 << p->nPageSize)' failed at src/sat/bsat/satClause.h:307

muchang, 2020-12-17:
ABC throws out an assertion failure when solving the following AIGER file using pdr.
[pdr.aig.zip](https://github.com/berkeley-abc/abc/files/5707564/pdr.aig.zip)
Here is the trace:
```
$ ./abc -c "read trigger.aig;pdr" 
ABC command line: "read trigger.aig;pdr".

abc: src/sat/bsat/satClause.h:307: Sat_MemAppend: Assertion `nInts + 3 < (1 << p->nPageSize)' failed.
Aborted
```

Operating System: Ubuntu 18.04
Commit ID: 73dcdab

alanminko, 2020-12-17:
It appears that this test case leads to unusually large learned clauses 
(something we have not seen so far).  Please feel free to change the 
parameter, which determines the buffer size in line 1140 of file 
"src/sat/bsat/satSolver.c".  Currently it is

>     Sat_MemAlloc_(&s->Mem, 17);
please feel free to try a larger value 18, 19, or 20. Incrementing it by 
one will double the memory used by the SAT solver, but it may be worse 
it to avoid the crash.

muchang, 2020-12-18:
Thanks! We will try to play with buffer size and see.