berkeley-abc/abc #116: Unknown platform error on macOS Big Sur (11.1) on AArch64

sunipkmukherjee, 2020-12-21:
abc_global.h platform checks fail to define ABC_PTRDIFF_T etc, I had to manually patch this file to set the adequate lengths to get abc to build.

sterin, 2020-12-26:
Unfortunately, we don't have access to a machine running macOS on Apple Silicon.  Could you provide some additional information?

1. What command do you use to build ABC?
2. During build, the makefile creates a small executable called `arch_flags`. Could you run it and report the here what output it generates?
3. Could you provide the error messages produced by the compiler?

Thanks,
Baruch

sunipkmukherjee, 2020-12-28:
I have a few insights to provide. I will detail what I did the first time:

1.  I tried to use cmake to build ABC. I have a homebrew installation for aarch64, containing cmake etc. In the command line when I execute cmake .. (in a build directory under abc) I initially get the following message with an error:
`-- The C compiler identification is AppleClang 12.0.0.12000032
-- The CXX compiler identification is AppleClang 12.0.0.12000032
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
./arch_flags.c:1:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
-- Configuring done
-- Generating done
-- Build files have been written to: /path/to/abc/build`

It is noteworthy that `cmake` (aarch64) fails to configure abc build system. 

Executing `make` in such state generates a bunch of errors of the form `/path/to/abc/src/misc/util/abc_global.h:162:5: error: unknown platform
   #error unknown platform

/path/to/abc/src/misc/util/abc_global.h:311:78: error: use of undeclared
      identifier 'ABC_PTRINT_T'
  ...int      Abc_Ptr2Int( void * p )                { return (int)(ABC_PTRIN...`

and so on.

I was trying to debug with the CMakeLists.txt file in `vscode`, which is an Intel binary and I had a CMake package installed, which successfully configures the build system using cmake (because it is using an x86 version of cmake). If I execute `make` with these generated files, I get the `unknown platform` error as well, but I can patch them to have the correct size and proceed as if nothing is amiss. 

The `abc_arch_flags_program.exe` generated by the `vscode cmake` returns `-DLIN64 -DSIZEOF_VOID_P=8 -DSIZEOF_LONG=8 -DSIZEOF_INT=4`, and is an intel binary (confirmed by forcing it to run in arm64 mode using `arch -arm64`.

2. I can compile the arch_flags.c program under both arm64 and x86_64 (by invoking clang with `arch -arm64` or `arch -x86_64`. In both cases, the outputs are identical; `-DLIN64 -DSIZEOF_VOID_P=8 -DSIZEOF_LONG=8 -DSIZEOF_INT=4`, which is also produced by the x86_64 cmake generated `abc_arch_flags_program.exe`.
3. However, executing `make` in `abc/` without bothering with cmake shows no issues and builds perfectly.

I hope that addresses your comments. I am open to testing any patch/fix you suggest.

sterin, 2020-12-29:
Could you verify the executable generated by executing `make` in `abc/` is indeed  AArch64? If so, there shouldn't be any problem building ABC.

However, under CMake, some suggestions on the Web point that this might be a problem with the Xcode installation. They suggest running Xcode at least once and installing the command line tools using `xcode-select --install`.

https://discourse.cmake.org/t/help-with-cc-clang-environment-and-cmake/2221/9
https://stackoverflow.com/questions/51761599/cannot-find-stdio-h

sunipkmukherjee, 2020-12-30:
The executable generated by `make` is indeed AArch64. I think I went through the CMake route the moment I saw the CMake file, which was dumb.

I do have the Xcode tools installed, in fact I preemptively ran `xcode-select --install`, that did not resolve the issue. I tried to link the `include` directory to `/usr/include`, but SIP prevents me from doing that manually... So I believe the issue is in `cmake` and not the `abc` build system. However I did not encounter this issue with any other `cmake` builds I have been doing.

I will leave the issue open, feel free to close it if you think it is resolved.

nils1603, 2022-01-17:
I am getting the same error on my macOS when using cmake (it works fine, when using make). 

Is it possible to pass the the parameter to cmake?

sterin, 2022-01-23:
I think this should be resolved now. It turns out the problem had nothing to do with Apple Silicon, but rather with macOS and Xcode. 

The CMakeLists.txt is an ugly hack that runs make to gather the list of source files and compilation flags. It provides make with the C and C++ compilers to use, but as it turns out that, on macOS, these also require the SDKROOT. The fix passes this value to make.

https://github.com/berkeley-abc/abc/commit/0a536417f6be75fcf26720767a61bd1f57a685b5#diff-1e7de1ae2d059d21e1dd75d5812d5a34b0222cef273b7c3a2af62eb747f9d20a

I've added automated build for CMake using GitHub actions, so hopefully these will continue to work.

closed 2022-01-23 by sterin