berkeley-abc/abc #120: cover -I 40 -P 5 produces a SOP node with 6 product terms

RobertBaruch, 2021-02-17:
I thought -P 5 specified that a maximum of 5 should be used? Here is a zip file showing the problem: [_tmp_yosys-abc-e6jYo7.zip](https://github.com/berkeley-abc/abc/files/5992560/_tmp_yosys-abc-e6jYo7.zip)

This script causes the problem (it is used in yosys):

```
read_blif _tmp_yosys-abc-J3ER0a/input.blif;
read_library _tmp_yosys-abc-J3ER0a/stdcells.genlib;
strash;
ifraig;
scorr;
dc2;
dretime;
strash;
dch -f;
cover -I 40 -P 5;
write_blif _tmp_yosys-abc-J3ER0a/output.blif
```

I also note that this script does not cause the problem:

```
read_blif _tmp_yosys-abc-J3ER0a/input.blif;
read_library _tmp_yosys-abc-J3ER0a/stdcells.genlib;
strash;
dretime;
cover -I 40 -P 5;
write_blif _tmp_yosys-abc-J3ER0a/output.blif
```

alanminko, 2021-02-18:
Unfortunately I cannot reproduce this problem on the stand-alone version 
of ABC.  For me, it works fine and produces a correct result.

In general "cover" has not been used much and could have a corner-case 
bug.  Please try to isolate it by saving the file before "cover" and 
make sure it shows on the stand-alone version of ABC.

Thanks,

Alan

RobertBaruch, 2021-02-18:
I compiled the latest ABC version, and the output still contains a size-6 SOP.

```
Running ABC command: /mnt/f/abc/abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_blif <abc-temp-dir>/input.blif
ABC: + read_library <abc-temp-dir>/stdcells.genlib
ABC: Entered genlib library with 13 gates from file "<abc-temp-dir>/stdcells.genlib".
ABC: + strash
ABC: + ifraig
ABC: + scorr
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
ABC: + dc2
ABC: + dretime
ABC: + strash
ABC: + dch -f
ABC: + write_blif tmp_blif.blif
ABC: + cover 40 5
ABC: + write_blif <abc-temp-dir>/output.blif
```

Here's the blif file before cover: [tmp_blif.zip](https://github.com/berkeley-abc/abc/files/5999733/tmp_blif.zip)

alanminko, 2021-02-18:
> ABC: + dch -f ABC: + write_blif tmp_blif.blif ABC: + cover -I 40 -P 5

Let us read the usage message:

> usage: cover [-IP num] [-sxvh]
>                  decomposition into a network of SOP/ESOP PLAs
>                  (this command is known to have bugs)
>         -I num : maximum number of inputs [default = 8]
>         -P num : maximum number of products [default = 8]
>         -s     : toggle the use of SOPs [default = yes]
>         -x     : toggle the use of ESOPs [default = no]
>         -v     : toggle printing verbose information [default = no]
>         -h     : print the command usage

-I is the number of inputs (can be up to 40 in your script)
-P is the number of products, or cubes in the SOP (can be up to 5)

The results seems to be correct according to this definition...

RobertBaruch, 2021-02-18:
Doesn't the first SOP in the output have 6 product terms, not 5?

```
# Benchmark "netlist" written by ABC on Wed Feb 17 18:25:41 2021
.model netlist
.inputs ys__n0 ys__n1 ys__n5 ys__n6 ys__n8 ys__n9 ys__n13 ys__n14 ys__n16 \
 ys__n17 ys__n20 ys__n21 ys__n23 ys__n24 ys__n29 ys__n31 ys__n32
.outputs ys__n30
.names ys__n0 ys__n1 ys__n8 ys__n9 ys__n16 ys__n17 ys__n29 ys__n31 ys__n32 \
 new_n19_
------1-- 0
--10----- 0
--01----- 0
10------- 0
-------10 0
----10--- 0
.names ys__n0 ys__n1 ys__n13 ys__n14 ys__n31 ys__n32 new_n20_
01---- 0
----01 0
--10-- 0
...
```

alanminko, 2021-02-20:
Just had a chance to look into this.  Unfortunately the problem appears 
to be deeply-rooted and it is hard to figure it out.  I would need to 
rewrite some code to make it work and this is currently not possible.

Sorry for being so unhelpful.  Is it possible to create a work-around?   
Please feel free to share with me what is the original task that needs 
to be solved.

RobertBaruch, 2021-02-20:
The original task is to take Verilog code (or in my case, RTLIL code generated via nMigen), run it through Yosys to generate a blif, pass it to ABC in order to create SOP cells, ultimately to post-process the result to try to fit the resulting logic onto a CPLD.

The specific file I showed was the logic for a 74688 8-bit equality comparator with output enable. The logic is essentially:

```
p_neq_q = n_oe ? 1 : (p != q)
```

My workaround is to write a tool that splits overly-large SOPs into smaller SOPs that chain.