berkeley-abc/abc #121: read a .genlib file containing LATCH

Flians, 2021-02-18:
Hi,
Why does the program crash when It reads a .genlib file containing LATCH?

```
C:\Users\Administrator\Desktop\ABC\ABC\reference>abc80930.exe
UC Berkeley, ABC 1.01 (compiled Sep 30 2018 23:38:01)
================== Command history ==================
source abc.rc
=====================================================
Loading resource file "abc.rc".
abc 01> r lib.genlib
Skipping latch "andf"...
Entered genlib library with 7 gates from file "lib.genlib".

C:\Users\Administrator\Desktop\ABC\ABC\reference>
```

And the content of the lib.genlib is followed as:
```
GATE not   1   dout=!din;                      PIN * NONINV 1 1 1 0 1 0
GATE and   1   dout=dina*dinb;                 PIN * NONINV 1 1 1 0 1 0
GATE cb    1   dout=dina+dinb;                 PIN * NONINV 1 1 1 0 1 0
GATE xor   1   dout=dina*!dinb+!dina*dinb;     PIN * NONINV 1 1 1 0 1 0
GATE dff   1   dout=din;                       PIN * NONINV 1 1 1 0 1 0
GATE zero   0   dout=CONST0;
GATE one    0   dout=CONST1;

LATCH andf 1 dout = dina + dinb*dout_NEXT;
PIN dina NONINV 1 1 1 0 1 0
PIN dinb NONINV 1 1 1 0 1 0
SEQ dout dout_NEXT RISING_EDGE
CONTROL clk 1 1 1 0 1 0
CONSTRAINT * 5.67 -0.59
```

alanminko, 2021-02-18:
Because sequential cells from the genlib library are not supported in ABC.

Flians, 2021-02-18:
Thank you. 
But .genlib contains LATCH. What tools can support it?