berkeley-abc/abc #123: Skipping line ".start_kiss", ABC don't support kiss format for Finite State Machine?

zerowe-seven, 2021-03-25:
I created a case from Berkeley Logic Interchange Format (BLIF) February 22, 2005

which is 

---------------------
.model
.inputs v0
.outputs v3.2
.latch [6] v1 0
.latch [7] v2 0
.start_kiss
.i 1
.o 1
.p 8
.s 4
.r st0
0 st0 st0 0
1 st0 st1 0
0 st1 st2 0
1 st1 st1 0
0 st2 st0 0
1 st2 st3 1
0 st3 st2 0
1 st3 st1 0
.end_kiss
.code st0 00
.code st1 11
.code st2 01
.code st3 10
.names v0 [6]
1 1
.names v0 v1 v2 [7]
-1- 1
1-0 1
.names v0 v1 v2 v3.2
101 1
.end
---------------------

But ABC skipping the line for kiss and crash out, so ABC don't support kiss format for Finite State Machine?

UC Berkeley, ABC 1.01 (compiled Feb  3 2021 23:53:57)
abc 01> read_blif 1.blif
Line 6: Skipping line ".start_kiss".
Line 7: Skipping line ".i 1".
Line 8: Skipping line ".o 1".
Line 9: Skipping line ".p 8".
Line 10: Skipping line ".s 4".
Line 21: Skipping line ".code st0 00".
Line 22: Skipping line ".code st1 11".
Line 23: Skipping line ".code st2 01".
Line 24: Skipping line ".code st3 10".
Segmentation fault

alanminko, 2021-03-25:
KISS table for FSM description cannot be read into ABC, but the older 
tool, SIS, can read it.

zerowe-seven, 2021-03-25:
Thanks Alanminko!

Do you where we can download SIS or MVSIS nowadays?

alanminko, 2021-03-25:
We can find SIS here 
https://ptolemy.berkeley.edu/projects/embedded/Alumni/pchong/sis.html

Please note that MVSIS does not support KISS format and many other 
features of SIS, but it can process multi-valued circuits.

AileenYuan, 2025-07-18:
> KISS table for FSM description cannot be read into ABC, but the older
> tool, SIS, can read it.
> […](#)

Does abc support the minimization and re-encoding of state machine states in other ways?