berkeley-abc/abc #127: Assertion error computing structural choices on EPFL design hyp.blif

macd, 2021-05-12:
The assertion is triggered both by `&dch` and `&dch -x`  The design can be found at https://github.com/lsils/benchmarks
```
UC Berkeley, ABC 1.01 (compiled May  1 2021 14:39:22)
abc 01> read_blif hyp.blif 
abc 02> &get; &st
abc 02> &dch
abc: src/opt/dar/darRefact.c:589: Dar_ManRefactor: Assertion `(int)Aig_Regular(pObjNew)->Level <= Required' failed.
Aborted (core dumped)
```

alanminko, 2021-05-12:
I suspect this has to do with "hyp.blif" having too many logic levels 
and being quite impractical (unlike any of the design circuits we have 
seen).

If it is a show-stopper, one idea is to comment out this assertion.  It 
does not impact correctness and just signals that logic level counting 
when wrong (likely by a small amount, such as one level).  Another idea 
is to not use "&dch" for deep logic.  Please try "&synch2" or 
lighter-scripts such as "&syn2" and "&syn3".

macd, 2021-05-12:
No it is definitely not a show stopper.  Just wanted to document any issues I might find so others can see it here too.  Thanks for the answer.

thm123, 2021-08-05:
Thanks for your comments, I encountered the same problem and debug it for some days with no answer. So I come the issue part for help. GOOD JOB ! @macd

wjrforcyber, 2023-09-10:
> I suspect this has to do with "hyp.blif" having too many logic levels and being quite impractical (unlike any of the design circuits we have seen). If it is a show-stopper, one idea is to comment [...]

Could this be avoid or is there a boundary of the level number that we could decide as "too many"?