berkeley-abc/abc #135: Cex for Abc9

CrazyKoE, 2021-08-31:
Hi there,

Got some questions about writing the counter-example of Abc9 Command. 

When I use chainbmc to resolve problem, the total number counter-example is shown perfectly. But when I tried to dump the counter-example file, it showed that NO CEX. Is there some way to dump the CEX for chainbmc and other kind like that?

emamajanyan, 2023-03-14:
Hi @CrazyKoE 
have you found a solutions to dump CEX ?
I've the similar situation - I need to write out the counter-example after &cec command, but write_cex reports an empty network.

```
abc - > &cec -v -w ./small/impl_m.aig ./small/spec.aig
Networks are NOT EQUIVALENT. Output 0 trivially differs (different phase).  Time =     0.00 sec
abc - > write_cex neg.cex
Empty network.
```

sterin, 2023-03-19:
If you use commands that start with and ampersand "&" then you should call the `&put` command before `write_cex`.

```
&put
write_cex neg.cex
```

emamajanyan, 2023-03-20:
> If you use commands that start with and ampersand `&` then you should call the `&put` command before `write_cex`.
> 
> ```
> &put
> write_cex neg.cex
> ```

Thanks for your comment @sterin 
Yes, it should be converted back to the old ABC network before `write_cex`.  I tried it as well, unfortunately the result is the same right after the `&put` command.
```
abc - > &cec ./small/impl_m.aig ./small/spec.aig
Networks are NOT EQUIVALENT. Output 0 trivially differs (different phase).  Time =     0.00 sec
abc - > &put
Error: Empty network.
``` 

I've also tried the `&cexinfo` which is an ABC9 command
```
abc - > &cec ./small/impl_m.aig ./small/spec.aig
Networks are NOT EQUIVALENT. Output 0 trivially differs (different phase).  Time =     0.00 sec
abc - > &cexinfo
Error: Abc_CommandAbc9CexInfo(): There is no AIG
```
I was expecting there should be a CEX when the Networks are NOT EQUIVALENT.

sterin, 2023-03-20:
Take the following with a grain of salt.

The counter example produced by`&cec` is with respect to the miter circuit. For the CEX to be available, you need to have the miter circuit as an old ABC network before running `&cec`. The `-m` flag to `&cec` tells it to use the current miter.

```
&r impl.aig
&miter spec.aig
&put
&cec -m
write_cex /dev/stdout
```

emamajanyan, 2023-03-22:
> Take the following with a grain of salt.
>
> The counter example produced by`&cec` is with respect to the miter circuit. For the CEX to be available, you need to have the miter circuit as an old ABC [...]

Thanks @sterin
Running the `&cec` and `write_cex` on an **old** ABC network produces the counter example.

sterin, 2023-03-22:
@emamajanyan Also, `&put` has a useful flag:

```
        -s     : toggle clearning verification status [default = yes]
```

That makes it possible to change the order a bit.

```
&r impl.aig
&miter spec.aig
&cec -m
&put -s
write_cex /dev/stdout
```