berkeley-abc/abc #137: How to do LUT-mapping with specified order of variables?

flpeng2014, 2021-09-13:
First, I use "bdd" to convert the circuit to BDD. 
Second, I use "order  variables_order.txt" to specify the order of variables.
Third, I use "if -K 6" or "lutmin -K 6" to do lut mapping. 
It seems that the second step doesn't have any effect on the lut mapping results? 
Dose there have anything wrong?

alanminko, 2021-09-14:
Command "order" changes the order of the primary inputs.  It is useful 
before command "collapse", which will use the new order to construct 
BDDs of the primary outputs in terms of primary inputs. Commands such as 
"if" and "lutmin" work with local functions of the nodes, which are not 
affected by the order of the primary inputs.

flpeng2014, 2021-09-16:
Thanks a lot! I think commands like "if" and "lutmin" may do lut mapping based on AIG. Please correct me if I am wrong.  What I expected is that when do lut mapping based BDD, the first selected bound set (the top k variables) will have largest logic levels and the last remained free set will have the smallest logic levels. Does ABC have command like this?

alanminko, 2021-09-16:
This is a special type of decomposition. ABC does not have it a separate 
command, but it can be implemented using ABC data structures.

flpeng2014, 2021-09-23:
Done

closed 2021-09-23 by flpeng2014

AlwaysWry, 2022-05-01:
Hi, I use the "if -K 6" to do the lut mapping,  but how can I check the information like the number of LUTs?  is there any command to print that? Thanks!