berkeley-abc/abc #148: How to use this tool to convert verilog to cnf?

rrlcs, 2021-12-31:
I could not find in the documentation that how to use this tool to convert a verilog into cnf format. It would be great if someone can help and update the documentation.

jitendrach94, 2021-12-31:
It may not be useful to use ABC to manipulate verilog file.
ABC work well when circuit data structure is in AIG format. 
However you can convert verilog file into AIG using Yosys tool then you can use ABC to generate CNF.

rrlcs, 2022-01-02:
> It may not be useful to use ABC to manipulate verilog file. ABC work well when circuit data structure is in AIG format. However you can convert verilog file into AIG using Yosys tool then you can use ABC to generate CNF.
> […](#)

Thanks, Jitendra.
Sorry, I had to post this as an issue as I could not find other ways to clarify.

closed 2022-01-02 by rrlcs