berkeley-abc/abc #165: Encountering If_ObjPerformMappingChoice: Assertion `pCutSet->nCuts > 0' failed.

tcal-x, 2022-04-20:
Potentially similar to #55.

This is the testcase:

[tcal_abc.zip](https://github.com/berkeley-abc/abc/files/8517818/tcal_abc.zip)

tcal-x, 2022-04-20:
Forgot to include the output:
```
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_lut <abc-temp-dir>/input.lut 
ABC: + read_box <abc-temp-dir>/input.box 
ABC: + &read <abc-temp-dir>/input.xaig 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =   5338/   5645  and =   20634  lev =   40 (3.22)  mem = 0.59 MB  box = 1002  bb = 6
ABC: + &scorr 
ABC: Warning: The network is combinational.
ABC: + &sweep 
ABC: + &dc2 
ABC: + &dch -f 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =   5338/   5645  and =   27594  lev =   29 (2.33)  mem = 0.65 MB  ch = 3973  box = 913  bb = 6
ABC: + &if -W 300 -v 
ABC: K = 5. Memory (bytes): Truth =    0. Cut =   52. Obj =  132. Set =  564. CutMin = no
ABC: Node =   27594.  Ch =  3456.  Total mem =    6.98 MB. Peak cut mem =    0.37 MB.
ABC: P:  Del = 5687.00.  Ar =   10398.0.  Edge =    29328.  Cut =   219994.  T =     0.05 sec
ABC: P:  Del = 5580.00.  Ar =   10137.0.  Edge =    28577.  Cut =   203822.  T =     0.04 sec
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 5526 has delay (3632.000000) exceeding the required times (3499.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 5539 has delay (4103.000000) exceeding the required times (3970.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 5743 has delay (3632.000000) exceeding the required times (3499.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 5755 has delay (4103.000000) exceeding the required times (3970.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 5883 has delay (3632.000000) exceeding the required times (3499.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 5896 has delay (4103.000000) exceeding the required times (3970.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 6100 has delay (3632.000000) exceeding the required times (3440.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 6116 has delay (4103.000000) exceeding the required times (3970.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 6206 has delay (4103.000000) exceeding the required times (3911.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26085 has delay (4574.000000) exceeding the required times (4441.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26088 has delay (4636.000000) exceeding the required times (4444.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26095 has delay (5267.000000) exceeding the required times (5047.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26107 has delay (5800.000000) exceeding the required times (5580.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26189 has delay (4574.000000) exceeding the required times (4444.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26191 has delay (5267.000000) exceeding the required times (5109.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26192 has delay (4574.000000) exceeding the required times (4441.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26199 has delay (5267.000000) exceeding the required times (5047.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26211 has delay (5800.000000) exceeding the required times (5580.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26219 has delay (4574.000000) exceeding the required times (4444.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26229 has delay (4574.000000) exceeding the required times (4441.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26231 has delay (5267.000000) exceeding the required times (5047.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26243 has delay (5800.000000) exceeding the required times (5580.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26248 has delay (4574.000000) exceeding the required times (4444.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26262 has delay (4574.000000) exceeding the required times (4441.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26264 has delay (5267.000000) exceeding the required times (5047.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26276 has delay (5800.000000) exceeding the required times (5580.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26871 has delay (5800.000000) exceeding the required times (5580.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26923 has delay (5800.000000) exceeding the required times (5580.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26938 has delay (5800.000000) exceeding the required times (5580.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 26964 has delay (5800.000000) exceeding the required times (5580.004883).
ABC: If_ObjPerformMappingAnd(): Warning! Node with ID 28715 has delay (5800.yosys-abc: src/map/if/ifMap.c:560: If_ObjPerformMappingChoice: Assertion `pCutSet->nCuts > 0' failed.
ABC: Aborted (core dumped)
```

davidar, 2024-04-01:
I've confirmed that applying a patch similar to https://github.com/berkeley-abc/abc/issues/55#issuecomment-546612665 to the `If_ObjPerformMappingChoice` function avoids the crash on this test case