berkeley-abc/abc #171: A pdr performance issue？

cuteDen-ECNU, 2022-05-13:
Hello, for the following benchmark, ABC pdr is much (more than 60 times) slower than IC3ref ( https://github.com/arbrad/IC3ref ), while it only has 135 And gates and 88 latches. As pdr is a better successor of IC3, does it trigger a performance issue in ABC?

```shell
$ time IC3ref/IC3 < case.aig
0

real    0m44.898s
user    0m43.353s
sys     0m1.447s
```

```shell
$ time abc/abc -c "pdr;write_cex -a 'log.cex'" case.aig
ABC command line: "pdr;write_cex -a 'log.cex'".

Invariant F[65] : 40162 clauses with 81 flops (out of 88) (cex = 0, ave = 40.15)
Verification of invariant with 40162 clauses was successful.  Time =     5.60 sec
Property proved.  Time =  3306.11 sec
Counter-example is not available.

real    55m11.752s
user    51m20.009s
sys     3m46.180s
```

We also use other tools like simplecar(https://github.com/lijwen2748/simplecar) to verify this case, and find that they are much faster than pdr on this case as well.

```shell
$ time ./simplecar -b -e case.aig ./

real    0m1.459s
user    0m1.405s
sys     0m0.051s
```

My system version and the version of abc:

```shell
$ cat /proc/version
Linux version 5.13.0-40-generic (buildd@ubuntu) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #45~20.04.1-Ubuntu SMP Mon Apr 4 09:38:31 UTC 2022

$ ./abc -h
UC Berkeley, ABC 1.01 (compiled May 11 2022 06:55:44)
```

cuteDen-ECNU, 2022-07-26:
Here are some examples where `pdr` performs much worse than `IC3ref`, do you have any ideas about why `pdr` is much slower than `ic3ref` on these cases?

```
time bin/IC3  < pdr/case1.aig
1

real    0m7.802s
user    0m7.070s
sys    0m0.053s

time bin/abc -c pdr;write_cex -a 'log.cex' pdr/case1.aig
ABC command line: "pdr;write_cex -a 'log.cex'".

Output 0 of miter "pdr/case1" was asserted in frame 189.  Time =   846.39 sec

real    15m34.485s
user    13m12.554s
sys    0m53.869s
```

```
time bin/IC3  < pdr/case2.aig
0

real    0m9.948s
user    0m8.647s
sys    0m0.170s

time bin/abc -c pdr;write_cex -a 'log.cex' pdr/case2.aig
ABC command line: "pdr;write_cex -a 'log.cex'".

Invariant F[76] : 32560 clauses with 100 flops (out of 108) (cex = 0, ave = 40.40)
Verification of invariant with 32560 clauses was successful.  Time =     3.41 sec
Property proved.  Time =  1697.57 sec
Counter-example is not available.

real    45m15.501s
user    25m49.483s
sys    2m28.141s
```

```
time bin/IC3  < pdr/case3.aig
1

real    0m23.172s
user    0m13.940s
sys    0m0.093s

time bin/abc -c pdr;write_cex -a 'log.cex' pdr/case3.aig
ABC command line: "pdr;write_cex -a 'log.cex'".

Output 0 of miter "pdr/case3" was asserted in frame 335.  Time =  1599.97 sec

real    40m48.963s
user    24m58.046s
sys    1m41.968s
```

```
time bin/IC3  < pdr/case4.aig
0

real    0m32.357s
user    0m22.902s
sys    0m0.336s

time bin/abc -c pdr;write_cex -a 'log.cex' pdr/case4.aig
ABC command line: "pdr;write_cex -a 'log.cex'".

[... 261 lines cut ...]
user    1m2.088s
sys    0m0.267s

time bin/abc -c pdr;write_cex -a 'log.cex' pdr/case18.aig
ABC command line: "pdr;write_cex -a 'log.cex'".

Output 0 of miter "pdr/case18" was asserted in frame 163.  Time =   979.49 sec

real    17m47.869s
user    15m19.575s
sys    0m59.974s
```

My system version and the version of abc:

```
cat /proc/version
Linux version 3.10.0-693.el7.x86_64 (builder@kbuilder.dev.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-16) (GCC) ) #1 SMP Tue Aug 22 21:09:27 UTC 2017

./abc -h
UC Berkeley, ABC 1.01 (compiled May 11 2022 06:55:44)
```

the case:
[pdr.zip](https://github.com/berkeley-abc/abc/files/9187217/pdr.zip)

sirandreww, 2024-03-12:
did you get an answer?

cuteDen-ECNU, 2024-03-13:
No, I haven't. I am still curious about the reasons for the issues.

Gy-Hu, 2025-04-29:
@cuteDen-ECNU Hi, I think you should treat `-nct` as a baseline. As this [paper](https://aman-goel.github.io/publications/goel2019empirical_preprint.pdf) mentions, `-nct` enables localization abstraction and category down, which is considered a baseline method nowadays.