berkeley-abc/abc #189: Error: Abc_CommandAbc9Strash(): There is no AIG

RorschachW, 2022-10-20:
[image]
why? pls

RorschachW, 2022-10-21:
> 

thank you very much.

wjrforcyber, 2023-08-12:
The answer from @weaversa is already very clear, I will attach a more detailed example from yosys abc mapping here:

```
#define ABC_COMMAND_LIB "strash; &get -n; &fraig -x; &put; scorr; dc2; dretime; strash; &get -n; &dch -f; &nf {D}; &put"
#define ABC_COMMAND_CTR "strash; &get -n; &fraig -x; &put; scorr; dc2; dretime; strash; &get -n; &dch -f; &nf {D}; &put; buffer; upsize {D}; dnsize {D}; stime -p"
#define ABC_COMMAND_LUT "strash; &get -n; &fraig -x; &put; scorr; dc2; dretime; strash; dch -f; if; mfs2"
#define ABC_COMMAND_SOP "strash; &get -n; &fraig -x; &put; scorr; dc2; dretime; strash; dch -f; cover {I} {P}"
#define ABC_COMMAND_DFL "strash; &get -n; &fraig -x; &put; scorr; dc2; dretime; strash; &get -n; &dch -f; &nf {D}; &put"
```

So as you can see, ABC command and ABC9 command can be used together, just before `&get` and after `&put`.

Ref: [Yosys abc mapping](https://github.com/YosysHQ/yosys/blob/master/passes/techmap/abc.cc)