berkeley-abc/abc #198: Does the saucy code carry a noncommercial-only license?

jamesjer, 2022-11-21:
The license terms for ABC are in copyright.txt, but copyright.txt is incomplete.  The portions of ABC written by people at Berkeley may be covered by the MIT-like license described there, but other software has been bundled into ABC, some of which has license terms that differ from those in copyright.txt.

In particular, `src/base/abci/abcSaucy.c` bundles saucy.  The upstream version has a non-free license due to a noncommercial-only use clause: http://vlsicad.eecs.umich.edu/BK/SAUCY/license-acad.html.  abcSaucy.c does not contain the license terms.  Instead, it refers to a file named LICENSE which does not exist in the ABC source tree.  Did the saucy developers convey this copy of saucy to ABC under the terms of the ABC license, or is it still under the noncommercial-only license, thereby causing ABC itself to be distributable only for noncommercial purposes?

copyright.txt should answer that question.  It doesn't.  If the license terms are in a file named LICENSE, then that should be in the source tree somewhere.  It isn't.

There are various other cases where the license files are included in subdirectories, but no mention is made in copyright.txt of the license obligations.  This includes the bundled versions of bzip2, cudd, satoko, and zlib, which have different license terms from those in copyright.txt.

I discovered all this while doing a license audit for the Fedora Linux distribution.  The saucy case is a real problem.  Please tell me whether we have acquired a noncommercial-only obligation or not.  If not, great!  But please say so in copyright.txt.  If we have, would it be possible to interface to bliss or nauty instead of saucy?  Both of them have free licenses.  Otherwise, we will have to remove ABC from Fedora.

Apteryks, 2025-05-02:
For what it's worth, Debian lists that file in its copyright file as:
```
Files: abc/src/base/abci/abcSaucy.c
Copyright: 2004 The Regents of the University of Michigan
License: University-of-Michigan
```

See: https://metadata.ftp-master.debian.org/changelogs//main/y/yosys/yosys_0.52-1_copyright

jamesjer, 2025-05-02:
FWIW, we have removed the abc package from Fedora, not just due to lack of response on this issue.  There has also been no response on reports of bugs that cause crashes.  We eventually decided that abc is effectively unmaintained and gave up on it.

Apteryks, 2025-05-04:
At least the yosys fork of abc seems maintained; it's included in Debian as `abc-yosys`; it seems Fedora has it under the `yosyshq-abc` name too.

jamesjer, 2025-05-04:
Yes, sorry, I should have mentioned that we switched to that fork.