berkeley-abc/abc #208: Some ABC9 commands can not run!!!!

sjyjytu, 2023-02-27:
When I compile the abc code as an binary, I find that some commands, such as &speedup, or some command with specific parameters, such as &if2 -m, can not work, saying that some assertion is not satisfied!
A case is as follows:
abc 01> &r /home/ei/workspace/yuanjianyong/project/HPPO2-LS/origin_files/priority.aig;&ps
priority : i/o =    128/      8  and =     978  lev =  250 (73.50)  mem = 0.02 MB
abc 01> &if -K 6;&speedup;&ps
abc: src/aig/gia/giaHash.c:133: Gia_ManHashStart: Assertion `*pPlace == 0' failed.
Aborted (core dumped)

alanminko, 2023-02-28:
Please try running "&st" (or some synthesis such as "&dch") before 
"&if".  Another idea, please try "&if -m -K 6" instead of "&if -K 6".   
If the assertion still fails, please include the test case.

sjyjytu, 2023-02-28:
Thank you for your kindly reply!
I have struggled to solve the problem today but still don't know why.
The problem I find is that if I use the command `make` or `make libabc.a` etc to compile the source code, it gets wrong with the operator "&speedup", but if and only if I use Clion to compile it in the Release mode, everything goes well!!
I think there is some problem with the compilation environment of ABC but I am not familiar with that.

alanminko, 2023-03-01:
It is likely that CLion in the release mode removes assertions. If it 
works fine and verifies, it may be ok to comment out this assertion in 
the original code.