berkeley-abc/abc #211: Question: generate sub-miters / partition the miter

emamajanyan, 2023-04-03:
Hi there,

I've some questions about generating and writing sub-miters.

I'm trying to partition the miter into sub-miter and for example I see the following in `cec' command help
```
-P num : partition size for multi-output networks [default = unused]                         
-p     : toggle automatic partitioning [default = no]  
```
so it seems there is such a functionality but I could not find a way to generate and write partitions as AIG files.

Is there a functionality in ABC that can generate sub-miters, i.e. partition the miter into sub-miters and write appropriate AIG files ?

sterin, 2023-04-09:
This can be easily done by a combination of the `miter` and `cone` commands. 

The `miter` command generates the miter, and the `cone` command replaces the current network with the cone of a PO (or more accurately a CO).

1. Generate the miter:
```
miter -c -P 2 xxx.aig yyy.aig
```

2. Extract the cone of the first (for example) partition:
```
cone -O 0
```

3. Write it to an AIG.
```
write_aiger miter_0.aig
```

emamajanyan, 2023-04-12:
Thanks @sterin.

I tried the commands that you've suggested, but got the same AIG for `cone -O 0` and `cone -O 1`.
i.e.
```
miter -c -P 2 xxx.aig yyy.aig
cone -O 0
write_aiger miter_0.aig
```
```
miter -c -P 2 xxx.aig yyy.aig
cone -O `
write_aiger miter_1.aig
```

in both cases the aig is
```
aig 0 0 0 1 0
0
c
```
I was expecting a different output, and was planning to save the current network after the `miter` command, then write the first AIG using `cone -O 0`, restore the network, write the second partition using `cone -O 1` and so on.
Maybe I missed something here, not sure.

The functionality that I'm trying to find is to able to partition the given miter into N parts.

There is some partitioning APIs used in `cec` command like `Abc_NtkPartitionNaive` or `Abc_NtkPartitionSmart`, and looks like we can use them to partition and then  write AIGs.

sterin, 2023-04-12:
@emamajanyan

I'm not sure there's a problem.

When constructing a miter, ABC rebuilds the two circuits on a common set of PIs/CIs. If the circuits are sufficiently similar (common in CEC tasks), many miter outputs get simplified to constant zero during structural hashing, i.e. the corresponding POs/COs are equivalent. 

Are you sure this is not what we're seeing here?

Unfortunately, I don't think `Abc_NtkPartitionSmart` is exposed directly as a command. If the limited partitioning available through `miter` is not sufficient, you'll have to do it in C/C++. It doesn't seem to compicated, though.

emamajanyan, 2023-04-17:
@sterin - yes the ' Abc_NtkPartitionSmart' or similar functions are not exposed directly as a command, thus an new command need to be added in ABC that uses those functions.

Thanks for you help.

wjrforcyber, 2023-04-24:
> 

Another thing that may need attention is, most of these partition functions like `Abc_NtkPartitionSmart` (there are a lot...) need an arg "partition size", which is not what you want, since it partitions the aig into parts with equal size...
I will try @sterin 's solution, I used to use DFS to get partition from every each one of the POs...