berkeley-abc/abc #216: abc's solver configuration for HWMCC'20?

anwu1219, 2023-04-27:
Hi there,

I'm writing to ask whether it is possible to share the solver configuration of abc for HWMCC'20? 

Thanks in advance!

sirandreww, 2024-02-07:
Did you get an answer to this?

sirandreww, 2024-02-19:
Do you have a guess as to how it was run?

sterin, 2024-02-21:
The complete configuration for HWMCC 2020 can be found at 

https://github.com/sterin/super-prove-build 

It's a wrapper project that collects all the relevant tools and build scripts. Checkout tag `hwmcc20-2` for the exact version used in the competition. There's a README.md file that describes how to build and run. It builds and runs on Ubuntu 18.04, later version might need some work, mainly because of the Python 2 to 3 transition.

sirandreww, 2024-02-23:
Thank you for the reference! From my understanding multiple engines were run at once for ABC results, as indicated by this:
```
engines = [
        ( 'pdr', ['abc', '-b', '-c', '&st; &put; pdr ; send_status'] ),
        ( 'bmc3', ['abc', '-b', '-c', '&st; &put; bmc3 -a'] ),
        ( ',bmc', ['bip', ',bmc', '-no-logo', '@@'] ),
        ( ',treb', ['bip', '-no-logo', ',treb', '@@'] ),
        ( ',treb -abs', ['bip', '-no-logo', ',treb', '-abs', '@@'] ),
    ]
```

GPJ-ZFF, 2024-10-20:
how to use bmc2 for an aig