berkeley-abc/abc #256: Mapping behavioral level Verilog file to MIG network

Maya7991, 2023-10-06:
I want to convert a behavioral level Verilog file to Majority-Inverter graph(MIG) network. The input is a Verilog file and the output MIG should also be written to Verilog file. Is it possible to achieve this task using ABC tool. I was not able to find any commands for this in the documentation.

This is an expected MIG representation of a 1-bit full adder:

```
module top(a, b, c, carry, sum);
  input a, b, c;
  output carry, sum;
  wire n1, n2;
    assign n1 = (a & b) | (b & c) | (c & a);
    assign n2 = (a & b) | (b & ~c) | (~c & a);
    assign carry = n1;
    assign sum = (c & ~n1) | (~n1 & n2) | (n2 & c);
endmodule
```

I am sorry for having to submit this as an issue but I couldn't find a solution even after a long search. I am also a beginner in this field which is why I am struggling with this question.

wjrforcyber, 2023-10-07:
As far as I know ABC focus on AIG optimisation, I think you could go for [mockturtle](https://github.com/lsils/mockturtle) for MIG representation and optimisation.

Maya7991, 2023-10-10:
@wjrforcyber but Mockturtle Lorina Verilog Parser has a limited parsing capabilities. You can check the issue#624 in mockturtle.
That's why I am looking for a solution in ABC.

phyzhenli, 2023-11-15:
You can use [Yosys](https://github.com/YosysHQ/yosys) to parse Verilog and output AIG/blif. Then use  [mockturtle](https://github.com/lsils/mockturtle) to read AIG/blif and convert it to MIG.

Maya7991, 2024-02-04:
> You can use [Yosys](https://github.com/YosysHQ/yosys) to parse Verilog and output AIG/blif. Then use [mockturtle](https://github.com/lsils/mockturtle) to read AIG/blif and convert it to MIG.

Thank you for your suggestion

closed 2024-02-04 by Maya7991