berkeley-abc/abc #257: "&mfs" step gives incorrect result on big endian for objects with >5 variables

zeldin, 2023-10-08:
This was originally reported against Yosys as YosysHQ/yosys#3993.

A [designed created with LiteX](https://github.com/pottendo/RVCop64-pottendo) containing a VexRiscv-SMP core with 64-bit databus was synthesized with the same yosys version on ppc64 and ppc64le, both with and without `abc9.nomfs`
With `abc9.nomfs` both ppc64 and ppc64 syntesize the same JSON, from which a working bitstream can be generated.
However, without `abc9.nomfs` different JSONs are generated on ppc64 and ppc64le, and only the one geneated by ppc64le can be turned into a working bitstream.

The reason is because `giaMfs.c` makes used of something referred to as "truth tables", which are really bit vectors.  Rather than using a portable opaque implementation of a bit vector, UCB have opted to create a non-portable non-opaque implementation.

The implementation uses a vector of limbs to represent a bit vector of arbitrary but constant length.  The limb containing a particular bit with number `n` is chosen as
```
limbs[n >> log2(limb_bits)]
```
To extract the bit from the limb by moving it to the LSB (as is done for example by `Abc_TtGetBit()`), an expression like
```
limb >> (n & (limb_bits-1))
```
is used.

So far this is standard fare.  The problem arises because the code accesses the same bit vector alternatingly with a limb size of 32 bits and 64 bits.  Because of how the indexing is constructed above, this gives a consistent view on little endian, but not on big endian.  Due to the periodic way small vectors are initialized, it seems to work by happy accident for vectors with 32 or less significant bits, but breaks on larger ones.

This can be fixed in two ways:

1.  Select one limb size to be the canonical limb size.
Accesses using this limb size can pick any indexing scheme, including the above, regardless of endianness.  However, accesses using the non-canonical limb size would need to take into account that the ordering of 32 bit limbs within 64 bit limbs differs between big and little endian.  Bit-parallel operations like AND and OR can ignore this difference (except for the special case of a vector consisting only of a single 32-bit limb, although this case could be eliminated by always allocating at least 64 bits), but code doing shifts of containing literal limb values which are not 32-bit periodic would need to be updated.

or

2. Modify the indexing depending on endianness to preserve a consistent view over different limb sizes.
There is of course nothing magical about little endian that makes the switching of limb size work, it's just that the indexing scheme has been constructed to make it work on little endian specifically.  Since the limb index is `n >> log2(limb_bits)`, it means that the bit with number 0 is stored in the _first_  (index 0) limb.  In order for the indexing to be consistent over limb size, it is therefore important that the bit with number 0 is also stored in the _first_ bit of the limb.  On little endian this means the LSB, but on big endian this is the MSB.  So with this arrangement, on big endian the code to move bit n into the LSB of the limb would become
```
limb >> ((~n) & (limb_bits-1))
```
instead, while
```
limb << ((n) & (limb_bits-1))
```
would move bit n to the MSB of the limb.
With this approach code does not need to care about the limb size, but any and all code which does shifts or which contains non-palindromic literal limb values would need updating.

Regardless of which of these approaches is chosen, the lack of abstraction in the code makes updating it a daunting task.  A good first step would probably be to increase the use of the abstractions that do exist.  For example, `Extra_PrintHex()` could easily have used `Abc_TtGetHex()`, but does not.  There are also multiple instances of the same constant truth tables, like the truth table `s_Truths6` in `utilTruth.h` which exists in verbatim copies in at least `giaTruth.c` and `giaMfs.c`...

zeldin, 2025-12-01:
Well, no.  While I haven't taken this to bitstream, I can see that using yosys 0.59.1 (which carries said abc commit) I still get different JSONs on LE and BE unless I enable `nomfs`.

I'd also like to contend that the approach of "sprinkle the code with byteswaps until it seems to work" is usually not the best one, especially from a maintainability perspective.  Going forward I'd encourage you to use endian agnostic abstractions instead.

zeldin, 2025-12-24:
I fixed the SIGSEGV in #465.  With that fix I get the same result from my test input on both big and little endian.

For the record I still believe that this inplace swapping is a bad idea, especially from a maintainability perspective, and that it would be better to adopt one of the two approaches I suggested in first comment of this issue.  If you agree I could work on a PR to that effect...

heshpdx, 2025-12-24:
@zeldin I think that's a good idea, but I'm not the maintainer. I'm just a co-traveler!