berkeley-abc/abc #271: Abc_AigUpdateLevelR_int: Assertion `Abc_ObjIsNode(pNode)' failed for some commands

YiKangOY, 2024-01-13:
I, I have an int2float circuit, I transformed it in to aiger by yosys, then I want to synthesize it through abc.

When I try "rf" or "resub", it will trigger `Abc_AigUpdateLevelR_int: Assertion `Abc_ObjIsNode(pNode)' failed`

[image]
[image]

The aig
[abc_int2float.zip](https://github.com/berkeley-abc/abc/files/13925297/abc_int2float.zip)
 and verilog file processed by Yosys is attatched.

Any help or suggestions will be appreciated!
Thanks!

Outsider225, 2024-01-15:
I met the  same problem. I transform vtr and koios benchmarks into blif using Odin II, then run abc with command "read stereovision0.odin.blif; strash; resyn2; if -K 6; print_stats; write_hie stereovision0.odin.blif stereovision0.abc_no_clk.blif; time"

YiKangOY, 2024-01-17:
Thanks, I will give it a try.

wjrforcyber, 2024-01-23:
I believe this issue is discussed by a lot of folks from Yosys and OpenLane, checkout [this thread](https://github.com/YosysHQ/yosys/issues/4039).
Replacing all `rewrite/refactor` with `drw/drf (-l)` may solve the problem.

YiKangOY, 2024-02-06:
I guess we can follow @wjrforcyber 's suggest, try to use drw and drf.  Thanks! Closing this issue.

closed 2024-02-06 by YiKangOY