berkeley-abc/abc #277: NOR gate mapping

Maya7991, 2024-02-04:
From an input BLIF file, I want to generate a gate level Verilog file that has only NOR gates . Can someone help me with the commands to be followed for this in ABC

JiaxiangPan, 2024-03-27:
I think this repository might be able to help you: https://github.com/debjyoti0891/MAGICNetlistGen

SaintBuddha1009, 2025-06-05:
Hello @Maya7991 @JiaxiangPan could u guide me on how to do it? Did it work?

JiaxiangPan, 2025-06-05:
Of course, no problem. You can check out this GitHub repository: [https://github.com/debjyoti0891/MAGICNetlistGen](https://github.com/debjyoti0891/MAGICNetlistGen). Take a look at the shell scripts inside — they work by reading a specific technology library and utilizing the technology mapping algorithms from ABC to generate the results.

SaintBuddha1009, 2025-06-05:
I am able to generate a NOR NOT based .BLIF files using nor_2 and nor_4 genlibs. But then after doing that, inside the simple-magic tool, i am not able to run the simple.py file. I tried changing the genlibs already present inside simple-magic but i guess i was unsuccessful. 

Once the .BLIF file is generated how to use the new genlibs here to generate OUTPUT files using simple_main.py?

SaintBuddha1009, 2025-06-05:
Warning: The design contains 2 undefined object types interpreted as blackboxes:
inv1 (7)  nor2 (11)  
Hierarchy reader converted 18 instances of blackboxes.
Io_WriteVerilogPos(): Omitted 21 feedthrough nets from output list of module (e.g. new_n22).
Io_WriteVerilogPos(): Omitted 21 feedthrough nets from output list of module (e.g. new_n22).
Traceback (most recent call last):
  File "simple_main.py", line 48, in <module>
    main()
  File "simple_main.py", line 37, in main
    Z3_input = Netlist_to_Z3_latency_nor2.netlist_to_z3(abc_output_path)    
  File "/home/balaji/Downloads/SIMPLE-MAGIC-master/Netlist_to_Z3_latency_nor2.py", line 260, in netlist_to_z3
    OutputString = re.search("output (.*?);", data, re.DOTALL).groups(0)[0].replace(" ","").replace('\n','').split(',')
AttributeError: 'NoneType' object has no attribute 'groups'

i am getting such an error when i run the SIMPLE-MAGIC python script. What might have gone wrong?

JiaxiangPan, 2025-06-05:
I think you misunderstood the usage of SIMPLE MAGIC. Your input only needs one blif file to convert it into a NOR/NOT netlist, which is achieved by invoking abc within SIMPLE MAGIC's code, rather than you doing this step. I think you should take a look at this repository [https://github.com/RotemBenHur/SIMPLER-MAGIC] . @SaintBuddha1009

SaintBuddha1009, 2025-06-16:
@JiaxiangPan Hello how are you doing. I am facing another issue. Whenever I use YOSYS to generate a .BLIF file for my .V file, ABC is not able to parse due to format issues. Did u face a similar issue? May ik what YOSYS version is suitable? Mine is YOSYS 0.9