berkeley-abc/abc #288: Any good tutorial to use ABC

quantrpeter, 2024-04-18:
Hi.
   Any good tutorial to use ABC? any tutorial to break combination logic into LUTs?
thanks

jeras, 2024-07-27:
This might be the primary documentation, but for now I am just guessing, since it doesn't have a date. 
https://people.eecs.berkeley.edu/~alanmi/abc/
After checking again the documentation still links to an old (now nonexisting) bitbucket repository.

To ABC developers: please add links to documentation, googling ABC does not give good results, and what I could find is either very old or lacks a date. You could add some note stating the documentation is outdated.

wjrforcyber, 2024-10-08:
I maintain a list of commands with related paper in a repo, this might help.
https://github.com/wjrforcyber/ABCPaperCheck