berkeley-abc/abc #339: Compilation problem

ashionjy, 2024-10-08:
When using the static library compilation mode, because there is no readline library, delete the -lreadline command, and the demo can be generated normally. But running _demo i10.aig_ results in an error: demo:Command not found.

I wonder if this is because there is no linked readline library.

Here are the commands used in the compilation process:
$ make libabc.a
$ gcc -Wall -g -c demo.c -o demo.o
$ g++ -g -o demo demo.o libabc.a -lm -ldl -lpthread
$ demo i10.aig

sarnold, 2025-03-09:
If you're in the project directory on Linux/Darwin then you need a ./ in front: `./demo  i10.aig`

sarnold, 2025-03-09:
Also, if you're using the static lib the `demo` binary should be ~130 MBs in size.