berkeley-abc/abc #349: How to dump a choice network?

chestercc1997, 2024-12-12:
``
read /abc/32_32_S_SP_AR_BK_GenMul.aig;choice;ps; 
``
abc/32_32_S_SP_AR_BK_GenMul: i/o =   64/   64  lat =    0  and =  18822 (choice = 2782)  lev =304
``
write test.aig
``
``
read test.aig;ps
``
abc/test       : i/o =   64/   64  lat =    0  and =  11682  lev =304
``
Can the choice network be dumped and read back in ABC? It seems that the dumped AIG contains choice nodes, but when reading it back, the choices cannot be recognized. I also tried using GIA write and read, which seems to keep the choice networks. However, when transferring it back to the AIG manager with &put, there are errors. If it cannot be put back into the AIG manager, it seems that ASIC tech mapping cannot be done in GIA.

wjrforcyber, 2024-12-13:
Yes and no.
As you mentioned:
>but when reading it back, the choices cannot be recognised.

It is because the choices nodes(specifically speaking the candidates nodes, not the representative one with fanout) have no fanout, they are treated as dangling nodes, and are collected and cleaned up by:
https://github.com/berkeley-abc/abc/blob/6754da13f27e9a84f23c4dea50b9d204754c8af9/src/base/io/ioReadAiger.c#L584

If you comment out this line, you will get back your aiger with all dangling nodes(all choices candidates):
```
./abc -c "read_aiger i10.aig; strash; choice; ps; write_aiger i10w.aig"
ABC command line: "read_aiger i10.aig; strash; choice; ps; write_aiger i10w.aig".

i10                           : i/o =  257/  224  lat =    0  and =   3764 (choice = 541)  lev = 51
The node number before writing 3764
./abc -c "read_aiger i10w.aig; ps;"                                    
ABC command line: "read_aiger i10w.aig; ps;".

i10w                          : i/o =  257/  224  lat =    0  and =   3764  lev = 51
```
but you can't tell the relations between representative node and candidate nodes (As you can see, there's no `(choice = XXX)`).
The original relations are maintained by **pointers** 
https://github.com/berkeley-abc/abc/blob/6754da13f27e9a84f23c4dea50b9d204754c8af9/src/base/abc/abc.h#L145
in ABC (See Figure 4 in [this paper](https://people.eecs.berkeley.edu/~alanmi/publications/2005/tcad05_map.pdf) for a clear illustration), so after dumping all these nodes without fanout into file format, this information is lost, so I am wondering if the `write_aiger` forgets to **check and remove all choices candidates**, it only checks the Ntk type:
https://github.com/berkeley-abc/abc/blob/6754da13f27e9a84f23c4dea50b9d204754c8af9/src/base/io/ioWriteAiger.c#L655
and when you use `write_aiger` it dump choices candidates if you have them in your network, actually I don't see the point here. Maybe there's other application I don't know, let's just ask @alanminko for his opinion see if this is by intention or just a small typo missing case.

So yes, you could get back all your nodes, but no, you can't tell which candidate belongs to which representative node easily.

Further more, I think it's not totally impossible if you'd like to save the relations of the choices somewhere else before dumping by yourself, and you can recover it after read aiger back anyway.

chestercc1997, 2024-12-17:
Hi Wang, thanks for sharing. I am doing some testing.

It seems that in AIG, pData is used to maintain the linked list of all equivalent choices of each node.
https://github.com/berkeley-abc/abc/blob/6754da13f27e9a84f23c4dea50b9d204754c8af9/src/base/abc/abc.h#L145
in Aig_Man_t, pEquivs uses a linked list to represent the equivalent nodes corresponding to each node.
https://github.com/berkeley-abc/abc/blob/8ba3d9b91c3d529223e90bcab0e9094bc015eac7/src/aig/aig/aig.h#L132
By traversing the support nodes of each choice node, their representative (the logic cone of the root node) can be obtained.

And when reading back, the mapping of which AIG node each choice node belongs to is not written in the AIG file, which results in the issue you mentioned: they are dumped but cannot be recognized as choices when read back. 

I’m testing in GIA to see if this information will be dumped, and I’ve also found that there are still some issues that I’m currently fixing when covert GIA choice network into AIG network.
``
./abc -c "read i10.aig;ps;&get;&choice;&put;ps;"
``
it will have an error : 
``
abc: src/base/abci/abcDar.c:1227: Abc_NtkFromDarChoices: Assertion `pTemp->pData != NULL' failed.
Aborted (core dumped)
``
https://github.com/berkeley-abc/abc/blob/8ba3d9b91c3d529223e90bcab0e9094bc015eac7/src/base/abci/abcDar.c#L1226
It should be said that for each choice node, this assert must be satisfied—having pData. However, it is currently unclear why the assert fails.

If this assert is commented out, the program can run, but it can be observed that even after dumping back to an AIG, the information about the choice nodes still isn't saved. By turning the below test from 0 to 1, it can be seen that the Abc_AigNodeIsChoice check still doesn't succeed.
https://github.com/berkeley-abc/abc/blob/8ba3d9b91c3d529223e90bcab0e9094bc015eac7/src/base/abci/abcDar.c#L1241-L1254
I will continue to investigate where the issues lie. If there is any progress, I will provide continuous updates

chestercc1997, 2024-12-17:
Solved with:#350
it works now , read the dumped choice  GIA aiger and using &put could reconstruct the choice network