berkeley-abc/abc #359: using another type of input file

yychen561, 2025-01-17:
how can i use a DEF or a tech-mapped Verilog as input, decompose it to AIG representation(command "strash"?), and then use ABC to optimize it?

wjrforcyber, 2025-02-16:
Maybe write your own, theoretically speaking any boolean complete network and be done by defining the ISOP for each one of the functional representative element and write into a blif format, and then read into ABC.
Maybe check [how yosys does it](https://github.com/YosysHQ/yosys/blob/38f858374c0ed83a6ac3ea66715263af015d7974/passes/techmap/abc.cc#L968).

yychen561, 2025-02-16:
You meant transforming the DEF or Verilog file to a BLIF file? However, I prefer in - memory operations instead of file reads and writes.
Now, I am trying to construct a network in ABC using the Miniaig package according to my DEF. Do you have any comments on this approach? Thanks for your help!

alanminko, 2025-02-24:
MiniAig is a good way to go.  It can also be passed in memory using API Abc_FrameGiaInputMiniAig().