berkeley-abc/abc #363: the LUT library about if

gongyuzhangzhao, 2025-02-03:
I've always wanted to know if IF mapping uses a LUT library. If so, where can I find the default LUT library?

alanminko, 2025-02-12:
By default "if" and "&if" use unit-delay / unit-area LUT library, which is assumed by default.  When the user enters a given library, it overrides the default one.  Examples of LUT libraries are given in this paper: https://people.eecs.berkeley.edu/~alanmi/publications/2012/date12_lut.pdf

wjrforcyber, 2025-03-21:
Hi @gongyuzhangzhao , I think Alan's answer is already clear enough, could you please close the issue if there's no further question? 

If you are locating the exact code implementation of the default one, it is loaded when abc initialises "FPGA mapping group":

https://github.com/berkeley-abc/abc/blob/f5ac2d4bd3fc749825795f18c795513e2e85a3b1/src/map/if/ifCom.c#L55-L57

And of course, you could load your own by using command `read_lut`:
```
usage: read_lut [-vh]
                  read the LUT library from the file
        -v      : toggles enabling of verbose output [default = yes]
        -h      : print the command usage
                                                
                  File format for a LUT library:
                  (the default library is shown)
                                                
                  # The area/delay of k-variable LUTs:
                  # k  area   delay
                  1      1      1
                  2      2      2
                  3      4      3
                  4      8      4
                  5     16      5
                  6     32      6
```
There's also an intuitive format for external lut loading [here](https://people.eecs.berkeley.edu/~alanmi/publications/2010/fpga10_speedup.pdf) in figure 4.1.

closed 2025-03-21 by gongyuzhangzhao