berkeley-abc/abc #372: There is a problem reading the ASAP7.lib file!

gongyuzhangzhao, 2025-02-21:
I encountered an issue while using read and read_lib to read the asap7.lib file. The specific situation is as follows:

abc 01> read_lib /root/autodl-tmp/zzt/DATA/benchmark_lib/7nm.lib
Segmentation fault (core dumped)

I would like to ask what this issue might be, as I have been troubleshooting for a long time.

gongyuzhangzhao, 2025-02-21:
I also encounter the same issue when reading other lib files.

gongyuzhangzhao, 2025-02-21:
I was able to successfully read it using Yosys-ABC. Why does ABC encounter this problem

wjrforcyber, 2025-02-23:
I suggest a MRE(Minimal reproducible example), your description doesn't have any detail such as commit hash, OS , gdb log, or the file you tried to read, which leads to a "no-one-can-do-anything" situation.

wjrforcyber, 2025-03-01:
Hi there, I noticed something wrong with `read_lib` and made a PR, could you please test if this fix your issue?

gongyuzhangzhao, 2025-03-03:
yes thanks！

closed 2025-03-03 by gongyuzhangzhao