berkeley-abc/abc #376: Googletest: should only download if explicitly requested

hzeller, 2025-03-01:
Currently, Googletest is attempted to be downloaded unconditionally if tests are enabled

https://github.com/berkeley-abc/abc/blob/master/CMakeLists.txt#L116-L128

This is problematic for packaging, as

  * in packaging context,s, the build is not allowed to connect to the 'net 
  * it is preferred to use the system-provided library.

So I suggest to only do that on some explicit, say `-DABC_DOWNLOAD_GTEST`, default false; by default it should just use the default googletest available on the system (1.14.0 is old enough that it should be available on all distros. ) using the `include(GoogleTest)` macro in CMake.

I see if I can get to it.

hzeller, 2025-03-01:
Ah, just looking at git annotate... ohai @QuantamHD :)

QuantamHD, 2025-03-01:
Why not download by default, and turn off if you're trying to package it? IMO we should optimize for dev velocity over packaging velocity

hzeller, 2025-03-01:
IMHO is an antipattern. Developers who care do not want that, in particular the kind that like velocity :)
(also, it would always download stuff on next clean). If there should be some sort of rigid dependency, it should be a `third_party/` submodule of sorts (also an antipattern, but slightly less worse).

Anyway, independent of the default, we first need an option to actually be able to choose, before we can define its default :)
I suggest something like `-DABC_USE_SYSTEM_GTEST`.

sarnold, 2025-03-09:
This is in my (ignored) PR [here](https://github.com/sarnold/abc-fork/blob/0f27a93988c905a0ec67def4ecf37beb1afb8e63/CMakeLists.txt#L454)

hzeller, 2025-03-10:
I like that @sarnold - can you make pull request against berkely-abc ? Looks like your link points to a fork but not a PR.

wjrforcyber, 2025-07-14:
Hi @hzeller ! I think for the second point you mentioned 

> it is preferred to use the system-provided library.

You could add an option `FIND_PACKAGE_ARGS` mentioned [here](https://cmake.org/cmake/help/latest/guide/using-dependencies/index.html#fetchcontent-and-find-package-integration). I am not going to PR here since I don't think there's plan for extensively applying gtest for all the interfaces. But if you need it for your own implementation then adding this arg is what you need.