berkeley-abc/abc #377: save/load for LUT network

phyzhenli, 2025-03-03:
Hi all,

I tried using save/load commands after if, but abc output an error:

[image]

It seems that save/load commands only work for asic mapping:

[image]

So my question is: Does save/load work for LUT mapping or is there are some similar commands for LUT mapping?

wjrforcyber, 2025-03-06:
Try `backup` and `restore`, tiny example with network statistic:
```
./abc -c "read_aiger i10.aig; strash; ps; if; ps; backup; strash; drw; drf; ps; restore; ps"
ABC command line: "read_aiger i10.aig; strash; ps; if; ps; backup; strash; drw; drf; ps; restore; ps".

i10                           : i/o =  257/  224  lat =    0  and =   2675  lev = 50
i10                           : i/o =  257/  224  lat =    0  nd =   862  edge =   2870  aig  =  3382  lev = 17
i10                           : i/o =  257/  224  lat =    0  and =   2057  lev = 49
i10                           : i/o =  257/  224  lat =    0  nd =   862  edge =   2870  aig  =  3382  lev = 17
```

phyzhenli, 2025-03-06:
haha, thanks for your reply Jingren. There is a diference between `backup` and `save` because the latter `compare and save the best one`.

I add two commands `save_lut` and `load_lut` to solve my problem.

closed 2025-03-31 by phyzhenli